isaac-sim/IsaacLab #1850: [Question] Can "isaacsim.robot.surface_gripper" been used in parallel  RL?

Xu-Xuanhui, 2025-02-13:
### Question

I saw the dependency "isaacsim.robot.surface_gripper" was introduced in "[isaaclab.python.kit](https://github.com/isaac-sim/IsaacLab/blob/main/apps/isaaclab.python.kit)". Can "isaacsim.robot.surface_gripper" be used in parallel reinforcement learning? I didn't find a demo.

RandomOakForest, 2025-02-14:
Thank you for posting this. There are no known issues, so this should be possible. Some [recommendations for training in parallel are offered here](https://isaac-sim.github.io/IsaacLab/main/source/overview/reinforcement-learning/training_guide.html#training-with-parallel-environments).

Xu-Xuanhui, 2025-02-17:
Is there a demo related to "isaacsim.robot.surface_gripper"  which is a new API in isaaclab.

kellyguo11, 2025-02-24:
We are re-working the surface gripper API in Isaac Sim to support vectorization. It's not currently possible to use it in a parallelized fashion.

closed 2025-03-21 by RandomOakForest

Xu-Xuanhui, 2025-03-31:
> We are re-working the surface gripper API in Isaac Sim to support vectorization. It's not currently possible to use it in a parallelized fashion.

Xu-Xuanhui, 2025-03-31:
Is there any API currently available to simulate adsorption in a parallel training environment for reinforcement learning? Can adsorb and desorb.

Xu-Xuanhui, 2025-03-31:
> We are re-working the surface gripper API in Isaac Sim to support vectorization. It's not currently possible to use it in a parallelized fashion.

Is there any API currently available to simulate adsorption in a parallel training environment for reinforcement learning? Can adsorb and desorb.