isaac-sim/IsaacLab #1857: [Bug Report] Simulation program crashed when I change the name of task_env class.

chengfengke, 2025-02-14:
### Describe the bug

After I changed the name of the Task_Env class, when using GUI simulation, if I click on an asset within the display box and an arrow indicating displacement appears, the entire simulation program will crash.

### Steps to reproduce

[image]

[image]

[image]

[image]

All the things in FactoryEnvPick and FactoryEnv are totally same. I only changed the name of FactoryEnv to FactoryEnvPick, I realize that the error is torch._C._LinAlgError: linalg.inv, and it burnt out when I was trying to click one of the assests in the environment(I would act normally when I click nothing, even moving my camera). And I also met same error when I directly copied the same part of code on Isaac Sim 4.2 to 4.5. But I guess the inherent reason are not same.
### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version:4.5
- OS: Ubuntu 20.04
- GPU: RTX4090
- CUDA: 11.8
- GPU Driver: 

### Additional context

Add any other context about the problem here.

### Checklist

- [√ ] I have checked that there is no similar issue in the repo (**required**)
- [√ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1 Let the Simulation being activated by my customized class without the error.

RandomOakForest, 2025-02-14:
Thank you for posting this. This is likely a bug, we will investigate. Could you please provide the driver you are using? Thanks.

chengfengke, 2025-02-15:
[image]
thx a lot for your fast reply, My current driver version is 550.67.

KumoLiu, 2025-04-24:
I met the same `torch._C._LinAlgError`  when upgrading the isaacsim from 4.2 to 4.5 with the same setting. When the error occurs, Franka appears to be in a very unusual position as shown below.

[image]