isaac-sim/IsaacLab #1860: [Proposal] Event-Aware State Tracking in RigidObjectCollection

louislelay, 2025-02-14:
### Proposal

Introduce an event-aware state tracking feature in `RigidObjectCollection` by adding a `last_written_state` attribute. This property will store the most recent manually set object state (pose + velocity) while keeping it separate from default or system-managed states.  

The core capability is to automatically track when an object's state is explicitly modified (e.g., during events).

### Motivation

I'm working on a Jenga manipulation environment, and when developing a reward function, I needed an easy-to-use method to access an object's initial x and y pose for comparison.  

However, I realized that `default_object_state` was not reliable because, after a randomization event, it no longer reflected the actual initialized state of the environment.

### Additional context

- `last_written_state` is updated only when an object’s state is explicitly set (via `write_object_com_velocity_to_sim()` and `write_object_link_pose_to_sim()`).
- Does not interfere with physics updates, ensuring the separation of manual interventions from regular simulation dynamics.
- I have already implemented this feature and tested it successfully in my manipulation environment. If approved, I can submit a PR with the working code.  

### Checklist
- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

- [ ] `last_written_state` correctly updates only on manual changes (not automatic physics updates).  
- [ ] Users can retrieve `last_written_state` at any time to see the last explicitly set state.  
- [ ] The implementation does not interfere with the existing physics pipeline.  
- [ ] Documentation is updated.

RandomOakForest, 2025-02-14:
Thank you for your proposal. The team will review.

zoctipus, 2025-02-14:
I have had this need before as well! As an work around, I used a customized class EvenTerm that stores the init_pos and init_quat in the class field. Whenever the __call__ function is invoked, the init_pos and init_quat gets overriden by the new init_pos and init_quat.

Then for retrive init_pos and init_quat, I can do 

`env.event_manager.get_term_cfg("held_asset_material").init_pos`
`env.event_manager.get_term_cfg("held_asset_material").init_quat`

to access the init position and init_quat during each reset

I agree that this data retrival seems hacky, it will be more ideal if there is a cleaner way to do it like the one you proposed!

louislelay, 2025-02-18:
I should mention that this approach can also be generalized to both deformable and rigid objects. I was initially considering my specific use case, but the principle remains the same. 

It could also be extended to articulations, though I don't see as much relevance in that case compared to the others.

AntoineRichard, 2026-08-27:
Thanks @louislelay for the proposal, and for circling back to tell us where it landed.

Closing on your own 2025-08-06 note that the asset-data changes had resolved this for you. For anyone landing here later: `RigidObjectCollectionData` now exposes pose and velocity as separate lazy properties (`body_link_pose_w`, `body_com_vel_w` — the older `object_*` names still work but warn and are slated for removal in 4.0) rather than one concatenated state tensor. Those still track current sim state — there's no `last_written_state` persisting the reset-time pose, so please reopen if that specific accessor is what you need.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard