isaac-sim/IsaacLab #1863: Camera not working on GPU

cornuco-pia, 2025-02-14:
TypeError: can't convert cuda:0 device type tensor to numpy. Use Tensor.cpu() to copy the tensor to host memory first.

it crashes on 

rot_offset = convert_camera_frame_orientation_convention(
                rot, origin=self.cfg.offset.convention, target="opengl"
            )
            rot_offset = rot_offset.squeeze(0).numpy()

in Camera.__init__()

i use torch.set_default_device('cuda')
how do i fix this?

RandomOakForest, 2025-02-18:
Thanks for posting this. Could you share the versions of Isaac Sim and Isaac Lab you are using and the driver and OS you are on? Any steps to reproduce this issue that you can include would be helpful. Thanks.

cornuco-pia, 2025-02-18:
thanks for reaching out. specs:

[image]

to reproduce, you need to initialize camera with torch.set_default_device('cuda') or where simulation_app.device == 'cuda'. i suspect it's because sensor class takes its device directly from simulation_app (self._device = sim.device in SensorBase._initialize_imp()), otherwise i would be able to somehow operate camera on cpu.
the bug itself comes from Camera.__init__(): rot_offset = rot_offset.squeeze(0).numpy() where rot_offset tensor is converted to numpy without calling .cpu()

cornuco-pia, 2025-02-18:
isaacsim 4.5.0
isaaclab 2.0.0

cornuco-pia, 2025-02-24:
@RandomOakForest do you maybe have any updates about this issue?

kellyguo11, 2025-02-26:
It seems like torch.set_default_device('cuda') forced the torch default device to cuda, whereas the init code assumed that the `rot` tensor was created on CPU. We can update the code to force the CPU device to avoid the error.

cornuco-pia, 2025-02-28:
i fixed it by manually handling tensor allocation but i think it would be helpful if you'd updated the code. thank you for looking into this

PR #2002 (2025-03-05): Sets default tensor device to CPU for Camera rot buffer
# Description

In the Camera class, we assumed that the default device for the torch tensor is CPU when we allocate the `rot` buffer during init. This isn't always the case if the user overwrites the torch device beforehand. This fix explicitly sets the tensor device to CPU on allocation.

Fixes #1863 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with `./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documen [... cut]
commit 5235c278 (2025-03-05): Sets default tensor device to CPU for Camera rot buffer (#2002)

# Description

In the Camera class, we assumed that the default device for the torch
tensor is CPU when we allocate the `rot` buffer during init. This isn't
always the case if the user overwrites the torch device beforehand. This
fix explicitly sets the tensor device to CPU on allocation.

Fixes #1863 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.sh --format`
- [ ] I have made corresponding changes to th [... cut]
closed 2025-02-28 by cornuco-pia