isaac-sim/IsaacLab #1872: [Bug Report] Camera poses not updated for TiledCamera

sunsleaf, 2025-02-15:
Just a tiny bug, the in the [_update_buffers_impl](https://github.com/isaac-sim/IsaacLab/blob/ecf551f863a1638911c2c1851b2958b4be0d70e9/source/isaaclab/isaaclab/sensors/camera/tiled_camera.py#L230C9-L230C29)() function of TiledCamera class, a call to self._update_poses() seems to be missing. In the [Camera](https://github.com/isaac-sim/IsaacLab/blob/ecf551f863a1638911c2c1851b2958b4be0d70e9/source/isaaclab/isaaclab/sensors/camera/camera.py#L492) class, this is present. This leads to the data.pos_w attribute of TiledCamera class not being updated.

In [tiled_camera.py](https://github.com/isaac-sim/IsaacLab/blob/ecf551f863a1638911c2c1851b2958b4be0d70e9/source/isaaclab/isaaclab/sensors/camera/tiled_camera.py#L230C9-L230C29):
```
def _update_buffers_impl(self, env_ids: Sequence[int]):
    # Increment frame count
    self._frame[env_ids] += 1

    # Extract the flattened image buffer
    for data_type, annotator in self._annotators.items():
    ...
```

In [camera.py](https://github.com/isaac-sim/IsaacLab/blob/ecf551f863a1638911c2c1851b2958b4be0d70e9/source/isaaclab/isaaclab/sensors/camera/camera.py#L488):
```
def _update_buffers_impl(self, env_ids: Sequence[int]):
    # Increment frame count
    self._frame[env_ids] += 1
    # -- pose
    self._update_poses(env_ids)
    # -- read the data from annotator registry
    # check if buffer is called for the first time. If so then, allocate the memory
    if len(self._data.output) == 0:
    ...
```

I'm sorry if this is already the intended behavior. But after I added the call to _update_poses() in tiled_camera.py, the data.pos_w attribute gets correctly updated.

RandomOakForest, 2025-02-18:
Thanks for posting this. We will review and update if necessary.

hersh, 2025-05-01:
This seems to have been fixed. In https://github.com/isaac-sim/IsaacLab/blob/main/source/isaaclab/isaaclab/sensors/camera/tiled_camera.py, the code now looks like:

    def _update_buffers_impl(self, env_ids: Sequence[int]):
        # Increment frame count
        self._frame[env_ids] += 1

        # update latest camera pose
        if self.cfg.update_latest_camera_pose:
            self._update_poses(env_ids)