isaac-sim/IsaacLab #1879: [Bug Report] Failed to acquire interface: omni::physx::IPhysxFabric (pluginName: nullptr)

Andrii0607, 2025-02-15:
Following “Installing Isaac Lab” page I installed Isaac Lab to my virtual environment.
https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html#installing-isaac-lab

When I executed this command in my virtual environment to verify installation:
`python scripts/tutorials/00_sim/create_empty.py`

I got Physx related error: 

Traceback (most recent call last):
  File "/home/aaa/IsaacLab/scripts/tutorials/00_sim/create_empty.py", line 59, in <module>
    main()
  File "/home/aaa/IsaacLab/scripts/tutorials/00_sim/create_empty.py", line 42, in main
    sim = SimulationContext(sim_cfg)
  File "/home/aaa/IsaacLab/source/isaaclab/isaaclab/sim/simulation_context.py", line 260, in __init__
    super().__init__(
  File "/home/aaa/isaac_env/lib/python3.10/site-packages/isaacsim/exts/isaacsim.core.api/isaacsim/core/api/simulation_context/simulation_context.py", line 142, in __init__
    self._init_stage(
  File "/home/aaa/IsaacLab/source/isaaclab/isaaclab/sim/simulation_context.py", line 536, in _init_stage
    self._load_fabric_interface()
  File "/home/aaa/IsaacLab/source/isaaclab/isaaclab/sim/simulation_context.py", line 616, in _load_fabric_interface
    self._fabric_iface = get_physx_fabric_interface()
  File "/home/aaa/isaac_env/lib/python3.10/site-packages/omni/data/Kit/Isaac-Sim/4.5/exts/3/omni.physx.fabric-106.3.2+106.3.0.lx64.r.cp310.ub3f/omni/physxfabric/__init__.py", line 11, in get_physx_fabric_interface
    return _get_interface(get_physx_fabric_interface, acquire_physx_fabric_interface)
  File "/home/aaa/isaac_env/lib/python3.10/site-packages/omni/data/Kit/Isaac-Sim/4.5/exts/3/omni.physx.fabric-106.3.2+106.3.0.lx64.r.cp310.ub3f/omni/physxfabric/__init__.py", line 7, in _get_interface
    func.iface = acq()
RuntimeError: Failed to acquire interface: omni::physx::IPhysxFabric (pluginName: nullptr)

How to resolve this error?

System Info
    Isaac Sim Version:4.5
    OS: Ubuntu 22.04
    GPU: RTX2060
    Driver Version: 550.120

Lr-2002, 2025-02-16:
yeah. same as you , We need more help ! 
Isaac Sim: 4.5
OS: Ubuntu 20.04
GPU: RTX 4090

RandomOakForest, 2025-02-18:
Thank you for posting this. We will investigate.

deferrednebula, 2025-02-24:
Install a fresh copy of everything with pip. avoid installing the isaac sim with the download link

winster-bai, 2025-02-24:
Same error, how to download the right IPhysxFabric plugin extension version?

kellyguo11, 2025-02-24:
IPhysxFabric should be built in into the Isaac Sim package. could be something got corrupted or didn't download correctly when installing Isaac Sim. I would recommend trying a fresh install as recommended above.

closed 2025-04-18 by RandomOakForest