isaac-sim/IsaacLab #1880: [Bug Report] Bug regarding creating rigid object asset

hojae-io, 2025-02-16:
### Describe the bug

When I run some of the tutorial codes regarding (rigid) object spawning, I get those errors

```
2025-02-16 00:10:29 [6,858ms] [Error] [omni.ext._impl._internal] Failed to import python module isaaclab_tasks from /home/hjlee/git/IsaacLab.brl/IsaacLab/source/isaaclab_tasks. Error: 'NoneType' object has no attribute 'GetPrimAtPath'. Traceback:
Traceback (most recent call last):
  File "/home/hjlee/anaconda3/envs/isaaclab/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/_internal.py", line 187, in _custom_importer
    return _import_public(ext_module.path, ext_module.name, reload_enabled)
  File "/home/hjlee/anaconda3/envs/isaaclab/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/_internal.py", line 94, in _import_public
    module = import_module(module_name)
  File "/home/hjlee/anaconda3/envs/isaaclab/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/custom_importer.py", line 76, in import_module
    return importlib.import_module(name)
  File "/home/hjlee/anaconda3/envs/isaaclab/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/hjlee/git/IsaacLab.brl/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/__init__.py", line 25, in <module>
    from .utils import import_packages
  File "/home/hjlee/git/IsaacLab.brl/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/__init__.py", line 9, in <module>
    from .parse_cfg import get_checkpoint_path, load_cfg_from_registry, parse_env_cfg
  File "/home/hjlee/git/IsaacLab.brl/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/parse_cfg.py", line 16, in <module>
    from isaaclab.envs import DirectRLEnvCfg, ManagerBasedRLEnvCfg
  File "/home/hjlee/git/IsaacLab.brl/IsaacLab/source/isaaclab/isaaclab/envs/__init__.py", line 59, in <module>
    from extensions.humanoid.task import *
  File "/home/hjlee/git/IsaacLab.brl/extensions/humanoid/task/__init__.py", line 8, in <module>
    from . import humanoid_vanilla_runner_cfg, humanoid_vanilla_task_cfg
  File "/home/hjlee/git/IsaacLab.brl/extensions/humanoid/task/humanoid_vanilla_task_cfg.py", line 45, in <module>
    class MySceneCfg(InteractiveSceneCfg):
  File "/home/hjlee/git/IsaacLab.brl/extensions/humanoid/task/humanoid_vanilla_task_cfg.py", line 130, in MySceneCfg
    cone_object = RigidObject(cfg=cone_cfg)
  File "/home/hjlee/git/IsaacLab.brl/IsaacLab/source/isaaclab/isaaclab/assets/rigid_object/rigid_object.py", line 57, in __init__
    super().__init__(cfg)
  File "/home/hjlee/git/IsaacLab.brl/IsaacLab/source/isaaclab/isaaclab/assets/asset_base.py", line 77, in __init__
    self.cfg.spawn.func(
  File "/home/hjlee/git/IsaacLab.brl/IsaacLab/source/isaaclab/isaaclab/sim/utils.py", line 251, in wrapper
    prim = func(prim_paths[0], cfg, *args, **kwargs)
  File "/home/hjlee/git/IsaacLab.brl/IsaacLab/source/isaaclab/isaaclab/sim/spawners/shapes/shapes.py", line 212, in spawn_cone
    _spawn_geom_from_prim_type(prim_path, cfg, "Cone", attributes, translation, orientation)
  File "/home/hjlee/git/IsaacLab.brl/IsaacLab/source/isaaclab/isaaclab/sim/spawners/shapes/shapes.py", line 260, in _spawn_geom_from_prim_type
    if not prim_utils.is_prim_path_valid(prim_path):
  File "/home/hjlee/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/exts/isaacsim.core.utils/isaacsim/core/utils/prims.py", line 78, in is_prim_path_valid
    return get_prim_at_path(prim_path, fabric=fabric).IsValid()
  File "/home/hjlee/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/exts/isaacsim.core.utils/isaacsim/core/utils/prims.py", line 51, in get_prim_at_path
    return get_current_stage(fabric=fabric).GetPrimAtPath(prim_path)
AttributeError: 'NoneType' object has no attribute 'GetPrimAtPath'

Exception ignored in: <function AssetBase.__del__ at 0x72031408c430>
Traceback (most recent call last):
  File "/home/hjlee/git/IsaacLab.brl/IsaacLab/source/isaaclab/isaaclab/assets/asset_base.py", line 110, in __del__
    if self._initialize_handle:
AttributeError: 'RigidObject' object has no attribute '_initialize_handle'
2025-02-16 00:10:29 [6,858ms] [Error] [carb.scripting-python.plugin] Exception: Extension python module: 'isaaclab_tasks' in '/home/hjlee/git/IsaacLab.brl/IsaacLab/source/isaaclab_tasks' failed to load.

At:
  /home/hjlee/anaconda3/envs/isaaclab/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/_internal.py(213): startup
  /home/hjlee/anaconda3/envs/isaaclab/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/_internal.py(328): startup_extension
  PythonExtension.cpp::startup()(2): <module>
  /home/hjlee/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py(427): _start_app
  /home/hjlee/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py(212): __init__
  /home/hjlee/git/IsaacLab.brl/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py(638): _create_app
  /home/hjlee/git/IsaacLab.brl/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py(114): __init__
  /home/hjlee/git/IsaacLab.brl/IsaacLab/scripts/tutorials/01_assets/run_rigid_object.py(31): <module>

2025-02-16 00:10:29 [6,858ms] [Error] [omni.ext.plugin] [ext: isaaclab_tasks-0.10.22] Failed to startup python extension.
```
I think the error is regarding asset spawing (prim_utils, create_prim etc)
I check with orbit (old code) and it works without any errors.
So I guess the bug was introduced during IsaacLab update.

### Steps to reproduce

`python IsaacLab/scripts/tutorials/01_assets/run_rigid_object.py`
or
`python IsaacLab/scripts/tutorials/03_envs/create_cube_base_env.py `

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 4.5.0
- OS: Ubuntu 22.04
- GPU:4070 Ti
- CUDA: 12.4
- GPU Driver: 550.120

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-02-18:
Thank you for posting this, we will review. Could you share which steps/doc you followed to install Isaac Lab? Thanks

hojae-io, 2025-02-19:
I followed this: https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html

Mayankm96, 2025-02-20:
I can't reproduce this issue.

closed 2025-03-21 by RandomOakForest