isaac-sim/IsaacLab #1881: [Question] ModuleNotFoundError: No module named 'isaacsim.core'

theAayushbajaj, 2025-02-16:
### Question

Evidently, `isaacsim==4.5.0` exists (all installed via pip: https://docs.isaacsim.omniverse.nvidia.com/latest/installation/install_python.html) and trying to the the `source\isaaclab_tasks\isaaclab_tasks\manager_based\manipulation\lift\config\franka\joint_pos_env_cfg.py` file that throws an error from `source\isaaclab\isaaclab\assets\articulation\articulation.py`.  Traceback:

`(isaaclab) PS C:\IsaacLab\IsaacLab> .\isaaclab.bat -p .\source\isaaclab_tasks\isaaclab_tasks\manager_based\manipulation\lift\config\franka\joint_pos_env_cfg.py
[INFO] Using python from: C:\Users\Ush\miniconda3\envs\isaaclab\python.exe
Traceback (most recent call last):
  File "C:\IsaacLab\IsaacLab\source\isaaclab_tasks\isaaclab_tasks\manager_based\manipulation\lift\config\franka\joint_pos_env_cfg.py", line 6, in <module>
    from isaaclab.assets import RigidObjectCfg
  File "C:\IsaacLab\IsaacLab\source\isaaclab\isaaclab\assets\__init__.py", line 41, in <module>
    from .articulation import Articulation, ArticulationCfg, ArticulationData
  File "C:\IsaacLab\IsaacLab\source\isaaclab\isaaclab\assets\articulation\__init__.py", line 8, in <module>
    from .articulation import Articulation
  File "C:\IsaacLab\IsaacLab\source\isaaclab\isaaclab\assets\articulation\articulation.py", line 16, in <module>
    import isaacsim.core.utils.stage as stage_utils
ModuleNotFoundError: No module named 'isaacsim.core'`

Something wrong with the installation?

sunzhon, 2025-02-17:
Got same error when run isaaclab as: 

File "/home/admin-1/workspace/IsaacLab/source/isaaclab/isaaclab/assets/articulation/articulation.py", line 16, in <module>
    import isaacsim.core.utils.stage as stage_utils
ModuleNotFoundError: No module named 'isaacsim.core'

sunzhon, 2025-02-17:
I fixed this issue via importting the package later.  The error happen probably because starting the SimulationApp() before importing the package "isaacsim.core". refer to this : https://forums.developer.nvidia.com/t/no-module-named-omni-isaac-core-after-pip-install-isaacsim/296307

WangYCheng23, 2025-02-18:
I got this error "Unable to expose 'isaacsim.simulation_app' API: libcarb.so: cannot open shared object file: No such file or directory"

PinkPanther-ny, 2025-03-05:
> Please only run scripts from the scripts/ folder, which are meant to be run as standalone scripts. These scripts ensure that SimulationApp is created before loading isaac sim and omniverse modules

why pycharm intellisense cannot find isaacsim.core?

csyyysc, 2025-04-01:
With issacsim installed in miniconda virtual env, and running `pip list`, it shows:

```
isaacsim                     4.5.0.0
isaacsim-app                 4.5.0.0
isaacsim-asset               4.5.0.0
isaacsim-benchmark           4.5.0.0
isaacsim-code-editor         4.5.0.0
isaacsim-core                4.5.0.0
isaacsim-cortex              4.5.0.0
isaacsim-example             4.5.0.0
isaacsim-extscache-kit       4.5.0.0
isaacsim-extscache-kit-sdk   4.5.0.0
isaacsim-extscache-physics   4.5.0.0
isaacsim-gui                 4.5.0.0
isaacsim-kernel              4.5.0.0
isaacsim-replicator          4.5.0.0
isaacsim-rl                  4.5.0.0
isaacsim-robot               4.5.0.0
isaacsim-robot-motion        4.5.0.0
isaacsim-robot-setup         4.5.0.0
isaacsim-ros1                4.5.0.0
isaacsim-ros2                4.5.0.0
isaacsim-sensor              4.5.0.0
isaacsim-storage             4.5.0.0
isaacsim-template            4.5.0.0
isaacsim-test                4.5.0.0
isaacsim-utils               4.5.0.0
```

When I tried to import submodule of isaacsim, it shows **Import "isaacsim.core.utils.prims" could not be resolved**. Yet, when running scripts, it does works and launch the simulator successfully.

What to adjust so that I can get code hints when using those pkgs? I have looked up the installation guide for several times, and still cannot find a solution.  

Also, I have adjust the **settings.json** in my .vscode folder like this:

```
{
    "python.defaultInterpreterPath": "/mypc/miniconda3/envs/myenv/lib/python3.10/site-packages/_isaac_sim/python.sh",
    "python.autoComplete.extraPaths": [
        "/mypc/miniconda3/envs/myenv/lib/python3.10/site-packages"
    ]
}
```

RandomOakForest, 2025-04-03:
Thank you for following up. I will close this issue for now. If you still face a related problem, please open a new issue question. Thanks.