isaac-sim/IsaacLab #1883: [Question] Viewport camera position and orientation

hojae-io, 2025-02-16:
Hi,
Is there anyway that I can get the viewport camera (default camera) position and orientation?

Thank you!

right-chan, 2025-02-17:
In my case, I was able to obtain the eye (camera position) and target (look-at position) using the following approach:

```python
from omni.kit.viewport.utility.camera_state import ViewportCameraState

env = gym.make("TaskName", env_cfg)
env = env.unwrapped
camera_prim_path = env.cfg.viewer.cam_prim_path
viewport_api = env.sim._viewport_context
prim = viewport_api.stage.GetPrimAtPath(camera_prim_path)

camera_state = ViewportCameraState(camera_prim_path, viewport_api)
eye = camera_state.position_world
target = camera_state.target_world

print(f"{eye=}, {target=}")
```

RandomOakForest, 2025-02-18:
Thanks for posting these. Let us know if you need further help.

closed 2025-03-21 by RandomOakForest