isaac-sim/IsaacLab #1887: [Proposal] Support Agent-Specific `state_space` in MAPPO

bikcrum, 2025-02-16:
### Proposal
Currently, in multi-agent reinforcement learning (MARL) environments, we can define **observation spaces** separately for each agent using a dictionary. However, the `state_space`, which serves as the **global input for centralized critics in MAPPO**, is assumed to be **shared across all agents**.

I would like to propose extending **MAPPO** to support **agent-specific `state_space` definitions**, enabling:
- Independent **value functions per agent** (like IPPO).
- More flexible **privileged critic inputs** separate from actor observations.
- Hybrid training where some agents use a centralized critic while others do not.

Example:
```python
@configclass
class ShadowHandOverEnvCfg(DirectMARLEnvCfg):
    possible_agents = ["right_hand", "left_hand"]
    action_spaces = {"right_hand": 20, "left_hand": 20}
    observation_spaces = {"right_hand": 157, "left_hand": 157}

    # Current (single shared state_space)
-   state_space = 290  

    # Proposed (agent-specific state_space)
+   state_space = {"right_hand": 290, "left_hand": 300}  
```

### Alternatives
- **Use IPPO instead of MAPPO**: This works for independent critics but does not allow privileged information in the critic while maintaining decentralized execution.
- **Manually concatenate different state spaces**: This is inefficient and requires unnecessary additional computation for agents that do not need a shared critic.
- **Modify policy architecture outside MAPPO**: Requires significant changes and breaks existing frameworks designed for centralized training with decentralized execution (CTDE).

### Additional Context
- This feature would enable **heterogeneous critic inputs**, improving learning efficiency and performance in **asymmetric multi-agent scenarios**.
- The current assumption that `state_space` is **identical for all agents** is too restrictive.
- This change should be **backward-compatible**, ensuring that if `state_space` is defined as a single value, it still works as expected.

### Checklist
- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria
- [x] Allow specifying `state_space` as a dictionary per agent.
- [x] Ensure MAPPO correctly handles different critic inputs for different agents.
- [x] Maintain backward compatibility with the existing single `state_space` setup.

RandomOakForest, 2025-02-18:
Thank you for posting this. The team will review.

StafaH, 2026-11-16:
Thanks @bikcrum for the proposal, and sorry it sat for so long.

Closing this as it's gone stale. skrl 2.x (which Isaac Lab now requires) reworked multi-agent state so MAPPO already keeps a separate state and critic input per agent, which makes Toni-SM/skrl#274 out of date. What's still missing is on the skrl side: its `IsaacLabMultiAgentWrapper` still passes one shared state to every agent, so that wrapper would need to support per-agent states first. If that lands upstream, feel free to reopen and we can look at how to expose it from `DirectMARLEnvCfg`.

closed 2026-11-16 by StafaH