isaac-sim/IsaacLab #1898: [Question] Legged Robot Foot Contact Penetratiom

evanzijianhe, 2025-02-18:
### Question

When running a trained policy of a legged robot, I noticed that one of its feet penetrates the ground. The corresponding toe motors output very large actions. However, this penetration does not occur when I open the USD file in the Isaac Sim GUI and stand the robot on the ground.

I would appreciate any suggestions for improving this behavior. Thank you!

[image]

https://github.com/user-attachments/assets/dd88baaa-850b-4c2e-8f26-d0590544ef63

### System Info
Describe the characteristic of your environment:

- Commit: IsaacLab 1.4.1
- Isaac Sim Version: 4.2.0
- OS: Ubuntu 22.04
- GPU:4090
- CUDA: 12.4
- GPU Driver: 535.183.01

```
  # general settings
  self.decimation = 4
  self.episode_length_s = 20.0
  self.sim.render_interval = self.decimation
  # simulation settings
  self.sim.dt = 1/500
  self.seed = 42
  self.sim.physx.solver_type = 1   #0: PGS (Projective Gauss-Seidel) 1: TGS (Truncated Gauss-Seidel)
  self.sim.physx.max_position_iteration_count = 4
  self.sim.physx.max_velocity_iteration_count = 1
  self.sim.physx.min_position_iteration_count = 4
  self.sim.physx.min_velocity_iteration_count = 1
  self.sim.disable_contact_processing = True
```

RandomOakForest, 2025-02-18:
Thank you for posting this. A similar question was asked earlier that may be of help, see comments to #1690. Let us know if that doesn't address your issue as well. Thanks.

KyleM73, 2025-02-20:
I couldn't tell from the video but when you initialize the scene is the robot above the ground? I have occasionally observed behavior like this if you initialize the robot such that there is already penetration at time step zero. Even if the robot is exactly on top of the ground, giving it an extra centimeter has sometimes resolved this for me. I'm not sure if that's your issue, but maybe something to try.

kellyguo11, 2025-02-24:
Another thing to look into is if the actions are too large for the robot, the simulator may have trouble solving for the collisions correctly. It might be worth considering adding some action constraints to avoid extremely large actions.

closed 2025-03-21 by RandomOakForest