isaac-sim/IsaacLab #1903: [Question] how can I lift the cube?

xiaolijz, 2025-02-19:
### Question

Hello, I'm encountering a problem with picking up a cube. When I close the gripper, I can't lift the cube. It seems that the relationship between the cube and the robot has not been established.                `robot.set_joint_position_target(gripper_close_pos, joint_ids=gripper_joint_ids) 
 scene.write_data_to_sim() 
   for _ in range(50): 
     sim.step().`

https://github.com/user-attachments/assets/05a0413a-36eb-43fd-8acc-1553bb304b9e

RandomOakForest, 2025-02-19:
Thanks for posting this. It may have to do with your friction settings. Could you share how you set up your robot config?

xiaolijz, 2025-02-19:
> Thanks for posting this. It may have to do with your friction settings. Could you share how you set up your robot config?

```
FRANKA_PANDA_CFG = ArticulationCfg(
    spawn=sim_utils.UsdFileCfg(
        usd_path=f"{ISAACLAB_NUCLEUS_DIR}/Robots/FrankaEmika/panda_instanceable.usd",
        activate_contact_sensors=False,
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            disable_gravity=False,
            max_depenetration_velocity=5.0,
        ),
        articulation_props=sim_utils.ArticulationRootPropertiesCfg(
            enabled_self_collisions=True, solver_position_iteration_count=16, solver_velocity_iteration_count=4
        ),
        collision_props=sim_utils.CollisionPropertiesCfg(collision_enabled=True,contact_offset=0.01, rest_offset=0.0),
    ),
    init_state=ArticulationCfg.InitialStateCfg(
        joint_pos={
            "panda_joint1": 0.0,
            "panda_joint2": -0.569,
            "panda_joint3": 0.0,
            "panda_joint4": -2.810,
            "panda_joint5": 0.0,
            "panda_joint6": 3.037,
            "panda_joint7": 0.741,
            "panda_finger_joint1": 0.04,
            "panda_finger_joint2": 0.04,
        },
    ),
    actuators={
        "panda_shoulder": ImplicitActuatorCfg(
            joint_names_expr=["panda_joint[1-4]"],
            effort_limit=87.0,
            velocity_limit=2.175,
            stiffness=80.0,
            damping=4.0,
        ),
        "panda_forearm": ImplicitActuatorCfg(
            joint_names_expr=["panda_joint[5-7]"],
            effort_limit=12.0,
            velocity_limit=2.61,
            stiffness=80.0,
            damping=4.0,
        ),
        "panda_hand": ImplicitActuatorCfg(
            joint_names_expr=["panda_finger_joint.*"],
            effort_limit=200.0,
            velocity_limit=0.2,
            stiffness=2e3,
            damping=1e2,
            friction=1.0,
        ),
    },
    soft_joint_pos_limit_factor=1.0,
)
"""Configuration of Franka Emika Panda robot."""

FRANKA_PANDA_HIGH_PD_CFG = FRANKA_PANDA_CFG.copy()
FRANKA_PANDA_HIGH_PD_CFG.spawn.rigid_props.disable_gravity = True
FRANKA_PANDA_HIGH_PD_CFG.actuators["panda_shoulder"].stiffness = 400.0
FRANKA_PANDA_HIGH_PD_CFG.actuators["panda_shoulder"].damping = 80.0
FRANKA_PANDA_HIGH_PD_CFG.actuators["panda_forearm"].stiffness = 400.0
FRANKA_PANDA_HIGH_PD_CFG.actuators["panda_forearm"].damping = 80.0
```
```
if current_goal_idx == 2:
                joint_pos[:, gripper_joint_ids] = gripper_close_pos.clone()
                robot.write_joint_state_to_sim(joint_pos, joint_vel)
                robot.reset()
                joint_pos = robot.data.joint_pos[:, robot_entity_cfg.joint_ids].clone()
                joint_pos_des = joint_pos[:, robot_entity_cfg.joint_ids].clone()
                print(f"Gripper position: {robot.data.joint_pos[:, robot_entity_cfg.joint_ids].clone()}")
            current_goal_idx = (current_goal_idx + 1) % len(ee_goals)
```

xiaolijz, 2025-02-20:
It also doesn't seem to work. 
`            if current_goal_idx == 2:
                for i in range(150):
                    robot.set_joint_position_target(gripper_close_pos, joint_ids=gripper_joint_ids)
                    scene.write_data_to_sim()
                    sim.step()
                    scene.update(sim_dt)`

StrainFlow, 2025-02-24:
The friction settings would be in the physics properties of both the gripper and the cube in their USD files. I'm fairly confident that the franka that comes with Isaac sim has friction values set, how are you creating the cube?

xiaolijz, 2025-02-25:
> The friction settings would be in the physics properties of both the gripper and the cube in their USD files. I'm fairly confident that the franka that comes with Isaac sim has friction values set, how are you creating the cube?

Thank you for your reply, I have solved this problem. Just add a line of code
`gripper_close_vec = torch.tensor([[-1.0,-1.0]],device=robot.device)
robot.set_joint_velocity_target(gripper_close_vec,joint_ids=gripper_joint_ids)
robot.set_joint_velocity_target(gripper_close_vec,joint_ids=gripper_joint_ids)`
 Previously, I only set the gripper joint angles without setting the speed.

xiaolijz, 2025-02-25:
> The friction settings would be in the physics properties of both the gripper and the cube in their USD files. I'm fairly confident that the franka that comes with Isaac sim has friction values set, how are you creating the cube?

Thank you for your reply, I have solved this problem. Just add a line of code
```
gripper_close_vec = torch.tensor([[-1.0,-1.0]],device=robot.device) 
robot.set_joint_position_target(gripper_close_pos, joint_ids=gripper_joint_ids)
robot.set_joint_velocity_target(gripper_close_vec,joint_ids=gripper_joint_ids)
```
Previously, I only set the gripper joint angles without setting the speed.

closed 2025-03-21 by RandomOakForest