isaac-sim/IsaacLab #1910: [Question] Does TGS stand for "Temporal Gauss-Seidel" or "Truncated Gauss-Seidel"?

AliBharwani, 2025-02-19:
In `class PhysxCfg`, the two options for `solver_type` are listed as PGS (Projective Gauss-Seidel) or TGS (Truncated Gauss-Seidel). This appears to be a typo as the PhysX documentation calls the TGS solver "Temporal Gauss-Seidel"

Toni-SM, 2025-02-19:
Hi @AliBharwani 

Please, see: https://nvidia-omniverse.github.io/PhysX/physx/5.5.1/docs/RigidBodyDynamics.html#projected-gauss-seidel-and-temporal-gauss-seidel

RandomOakForest, 2025-02-20:
Thank you for catching this. You are correct. We will fix.

Kyu3224, 2025-06-30:
Hi, I recently noticed the same thing while looking through the code and was planning to open a similar issue.
I believe the `PhysxCfg` defined in [`simulation_cfg.py`](https://github.com/isaac-sim/IsaacLab/blob/main/source/isaaclab/isaaclab/sim/simulation_cfg.py) needs to be updated.
Let me know if you'd like a PR for this fix.

PR #3011 (2025-07-30): Fix typos and docstring inconsistencies across environment and utility files
# Description

Fixes #1910,  #3010

### ✅ Typo Fixes

- Replaced `tome` → `time` in the following files:
  - `velocity_command.py`
  - `quadcopter_env.py`
  - `pre_trained_policy_action.py`
  - `pose_command.py`
  - `pose_2d_command.py`
  - `imu.py`
  - `contact_sensor.py`

- Corrected article usage `an numpy` → `a numpy` in:
  - `manager_based_rl_env.py`
  - `direct_rl_env.py`
  - `direct_marl_env.py`

- Removed repeated determiners (`the the`, `then then`) in:
  - `visualization_markers.py`

- Fixed incorrect expans [... cut]
commit 45c58bf4 (2025-07-30): Fix typos and docstring inconsistencies across environment and utility files (#3011)

# Description

Fixes #1910,  #3010

### ✅ Typo Fixes

- Replaced `tome` → `time` in the following files:
  - `velocity_command.py`
  - `quadcopter_env.py`
  - `pre_trained_policy_action.py`
  - `pose_command.py`
  - `pose_2d_command.py`
  - `imu.py`
  - `contact_sensor.py`

- Corrected article usage `an numpy` → `a numpy` in:
  - `manager_based_rl_env.py`
  - `direct_rl_env.py`
  - `direct_marl_env.py`

- Removed repeated determiners (`the the`, `then then`) in:
  - `visualization_markers.py`

- Fixed incorre [... cut]