isaac-sim/IsaacLab #1921: [Bug Report] Bug title

xinyu-66, 2025-02-21:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Hi, the UR5e robot end effector penetrate the rigid object.

### Steps to reproduce
**Load object to scene:**

base: RigidObjectCfg = RigidObjectCfg(
            prim_path="{ENV_REGEX_NS}/Base",
            init_state=RigidObjectCfg.InitialStateCfg(pos=[0.65, 0.0, 0.024], rot=[0.707, 0.707, 0, 0]),
            spawn=UsdFileCfg(
                usd_path="/home//Desktop/PiH_assets/plate.usd",
                scale=(0.001, 0.001, 0.001),
                rigid_props=RigidBodyPropertiesCfg(
                    solver_position_iteration_count=16,
                    solver_velocity_iteration_count=1,
                    max_angular_velocity=1e-6,
                    max_linear_velocity=0.1,
                    max_depenetration_velocity=1e-6,
                    disable_gravity=False,
                ),
            ),
        )

**ur5e configuration file:**
"""Configuration for the Universal Robots.

The following configuration parameters are available:

* :obj:`UR5e_peg_CFG`: The UR5e arm with a peg.

"""

import isaaclab.sim as sim_utils
from isaaclab.actuators import ImplicitActuatorCfg
from isaaclab.assets.articulation import ArticulationCfg
from isaaclab.utils.assets import ISAACLAB_NUCLEUS_DIR

##
# Configuration
##
UR5e_PEG_CFG = ArticulationCfg(
    spawn=sim_utils.UsdFileCfg(
        usd_path=f"/home/xinyu_sim45/Desktop/PiH_assets/ur5e_4.5/ur5e.usd",
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            disable_gravity=False,
            max_depenetration_velocity=5.0,
        ),
        articulation_props=sim_utils.ArticulationRootPropertiesCfg(
                enabled_self_collisions=True, solver_position_iteration_count=4, solver_velocity_iteration_count=0
        ),
        activate_contact_sensors=False,
    ),
    init_state=ArticulationCfg.InitialStateCfg(
        joint_pos={
            "shoulder_pan_joint": 0.0,
            "shoulder_lift_joint": -1.57,
            "elbow_joint": 1.57,
            "wrist_1_joint": -1.57,
            "wrist_2_joint": -1.57,
            "wrist_3_joint": 0.0,
        },
    ),
    actuators={
        "arm": ImplicitActuatorCfg(
            joint_names_expr=[".*"],
            velocity_limit=1.0,
            effort_limit=87.0,
            stiffness=800.0,
            damping=400.0,
        ),
    },
)

### System Info

Describe the characteristic of your environment:

- Commit: []
- Isaac Sim Version: [4.5.0]
- OS: [Ubuntu 22.04]
- GPU: [RTX 4090 Super]
- CUDA: [12.2]
- GPU Driver: [535.183.01]

### Additional context

[image]

[image]

[image]

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo. The test in Isaac Sim works well and collision is reasonable.

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] solution to make the collision to be reasonable without penetration

RandomOakForest, 2025-02-25:
Thank you for posting this. We will investigate.

xinyu-66, 2025-03-04:
Hi, if IsaacLab has `enable CCD(continuous collision detection)` option for rigid body?

supersglzc, 2025-03-06:
> Hi, if IsaacLab has `enable CCD(continuous collision detection)` option for rigid body?

You can enable it in simcfg. But it will make the simulation much slower.

AntoineRichard, 2026-08-27:
@calmdw appreciate the detailed report and the screenshots of the peg-in-hole penetration.

We can't reproduce this one: it depends on the private `PiH_assets/plate.usd` and custom `ur5e.usd` that weren't attached. The behavior is also consistent with the config — `max_depenetration_velocity=1e-6` effectively disables depenetration (assets in this repo use values from 1.0 up to 1000.0), and the 0.001 mesh scale compounds it. Enabling CCD or raising solver iterations, as discussed in the thread, is the usual remedy.

Closing for now — please reopen with a self-contained repro on a current release if it persists.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard