isaac-sim/IsaacLab #1927: [Question] particle rendering

NAMHAUK, 2025-02-24:
### Question

Basic questions, related to robot learning, that are not bugs or feature requests will be closed without reply, because GitHub issues are not an appropriate venue for these.

Advanced/nontrivial questions, especially in areas where documentation is lacking, are very much welcome.

For questions that are related to running and understanding Isaac Sim, please post them at the official [Isaac Sim forums](https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/isaac_sim_forums.html).

RandomOakForest, 2025-02-27:
Thank you for posting this question. We are taking a closer look into this to define what exactly is the issue.

ItsGJun, 2025-04-04:
I'm also having the same issue!

RandomOakForest, 2025-04-11:
Following up, is there any code you could share so we could reproduce this issue? Thanks

NAMHAUK, 2025-04-15:
My problem can simply be reproduced by proceeding to the next two steps.

1. Create particles using **particle sampler** in IsaacSim -> save as `water.usd` file
2. in IsaacLab, import to below code
```
water = sim_utils.UsdFileCfg(usd_path=f"mypath/water.usd")
water.func("/World/envs/env_0/water", water, translation=(0.0, 0.0, 0.0))
```

This is what I created particles **in IsaacSim.**
[image]

robegi, 2025-09-15:
Hi @NAMHAUK 

just wanted to report that the old method (issue https://github.com/isaac-sim/IsaacLab/issues/428) that I am testing on the latest versions of IsaacLab and Isaac Sim is still working. Don't know if you have noticed it, but I just wanted to report it.

Unfortunately the only way I found is still by using the CPU pipeline, and another solution does not seem to be available at the moment ([this](https://forums.developer.nvidia.com/t/particle-system-issues-with-fabric-enabled/309169/6) seems to confirm that).

Did you find a way to make it more efficient?

robegi, 2025-09-15:
@NAMHAUK yes, as far as I can tell. It is working or me. Also, remember to add these two lines when setting the scene up:

```python
physx_interface = acquire_physx_interface()
physx_interface.overwrite_gpu_setting(1)
```

If you did not previously do it, this is most likely the cause of your issues, so try and use these even in your current version. I believe that it is a persistent setting, so even if you delete these two line the gpu setting will still be overwritten if you executed just it once.

Yes I saw that the view API is missing for the particles. This is absurd since the internal buffers must have the particle's positions somewhere, but they are not accessible. Hope they fix it someday.