isaac-sim/IsaacLab #1928: [Bug Report] base_height_l2 Bug

xliu0105, 2025-02-24:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Using base_height_l2 in a manager-based reinforcement learning environment and using a raycaster sensor to adjust the target height based to account for the terrain, I found that with this setup, the reward value for this item would become '-inf' after about 80 iterations during training.

### Steps to reproduce

The settings for this item are: 
pnealize_base_hight = RewTerm(func = mdp.base_height_l2, weight = -0.9, params={"target_height":0.35, "sensor_cfg": SceneEntityCfg(name="raycaster")})

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 4.5
- OS: Ubuntu 22.04
- GPU: RTX 4060Ti 16G
- CUDA: 12.1
- GPU Driver: [e.g. 470.82.01, this can be seen by using `nvidia-smi` command.]

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-02-27:
Thank you for posting this. The team is already looking into a related issue (see #1698). We will investigate further.

AntoineRichard, 2026-09-01:
Hi @xliu0105, thanks for opening this issue! In this case, if you're getting -inf it's likely that the object is no longer above the terrain. I think the expected behavior is indeed to report -inf, letting the user know there was no hits. If you need something fancy that can deal with this, it's quite easy to build, look at #7343. 

Closing the issue for now, but let us know if you think this should be addressed.

closed 2026-09-01 by AntoineRichard