isaac-sim/IsaacLab #1929: [Bug Report] GUI Freezes After Minimization During Training

TheVaticanCameos, 2025-02-24:
### Describe the bug

When using Isaac Lab on Windows 11, minimizing the GUI during an ongoing training session and then restoring it causes the GUI to become unresponsive. The only functional buttons are the minimize, maximize, and close buttons in the top-right corner. All other UI interactions do not work. However, the training process continues normally, and outputs are still printed in the command-line terminal.

### Steps to reproduce

Here is an example.
1. Run the following command:
```
./isaaclab.bat -p scripts/tutorials/01_assets/run_articulation.py
```
2. Minimize the GUI while it is running.
3. Restore the GUI.
4. Try interacting with the GUI (e.g., adjusting settings, pausing).

### System Info

Describe the characteristic of your environment:

- Commit: 92f8930ff6283c664894b279464e1c9003325630
- Isaac Sim Version: 4.5.0
- OS: Windows 11
- GPU: RTX 4070 Super
- CUDA: 12.7
- GPU Driver: 565.90

### Additional context

Add any other context about the problem here.

### Checklist

- [√] I have checked that there is no similar issue in the repo (**required**)
- [  ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-02-27:
Thank you for catching this. We will follow up and aim to reproduce this.

kellyguo11, 2025-03-01:
I think this can be fixed by setting `skipWhileMinimized = true` in https://github.com/isaac-sim/IsaacLab/blob/main/apps/isaaclab.python.kit

closed 2025-04-18 by RandomOakForest