isaac-sim/IsaacLab #1932: [Question] Question title

BolunDai0216, 2025-02-24:
When running an IsaacLab environment, I get the following error message:

```bash
2025-02-24 22:53:46 [18,835ms] [Error] [rtx.materialdb.plugin] Texture upload failed for texture http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.5/NVIDIA/Materials/Base/Architecture/Shingles_01/Shingles_01_Normal.png
2025-02-24 22:53:46 [18,835ms] [Error] [rtx.materialdb.plugin] Failed texture loads reported for material usd::rtx_scope0::/World/ground/terrain/visualMaterial/Shader::::Z6Ahttp_3A_2F::Z58omniverse_2Dcontent_2Dproduction_2Es3_2Dus_2Dwest_2D2_2Eamazonaws_2Ecom::Assets::Isaac::ZD1_34_2E5::NVIDIA::Materials::Base::Architecture::Shingles_01::Shingles_01((11))
2025-02-24 22:53:46 [18,835ms] [Error] [omni.hydra] Failed wait for MDL material (Shingles_01(color,texture_2d,float,float,float,color,float,float,texture_2d,float,float,texture_2d,float,bool,texture_2d,float,texture_2d,bool,color,texture_2d,texture_2d,float,bool,texture_2d,float,bool,::base::mono_mode,float,float,float,texture_2d,float,texture_2d,bool,bool,bool,bool,int,float2,float,float2,float2,float,float2,float,float,bool)) snapshot
```

Due to this error I can see that the ground is red. Any idea what might be the issue?

Thanks in advance!

BolunDai0216, 2025-02-25:
I think the issue is that the Omniverse server is not installed, 

[image]

There should be a `localhost` under the `Omniverse` section. I was following the new installation procedure, which step should the Omniverse server be installed?

BolunDai0216, 2025-02-25:
After installing the Nucleus server using an older version of Omniverse launcher, the MDL issue got resolved, but now everytime I run IsaacLab I get the following warnings:

```bash
2025-02-25 14:44:30 [168ms] [Warning] [carb.omniclient.plugin]  3662543: OmniHub: ThreadId(03) Hub encountered error. Trying to reconnect to Hub. retry_reason="Hub failed to launch: Io(\"child exited with exit status: 101 without writing file \\\"/tmp/hub-bolun-8A69CD8E.config.json\\\"\")"
```

Any idea what this means? What is the suggested way to install the Nucleus server?

RandomOakForest, 2025-02-27:
Thanks for posting this. Could you elaborate on which steps you followed to install Isaac Lab and Isaac Sim? If possible, use the _Bug Report_ template. Regarding the last question above, this is related to issue #1777, and the team is working on it.

BolunDai0216, 2025-02-27:
@RandomOakForest Thanks for the response!

I was following the installation procedure detailed [here]( https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html).

RandomOakForest, 2025-04-03:
Thank you for following up. Is you still face this issue with the latest release, please submit a _Bug Report_. Will close this _question_ for now.

closed 2025-04-03 by RandomOakForest