isaac-sim/IsaacLab #1934: [Question] Modifying an existing Direct RL Environment with my robot usd file

zxfever, 2025-02-25:
In the `franka_cabinet_env.py` in the existing Direct RL Environment,  I change the franka usd file with my own robot file(converted by the urdf,about 50M size)

my setup:

- Isaac Sim Version: 4.2
- OS: Ubuntu 2204
- GPU: RTX 4090 
- CUDA: 12.4
- GPU Driver: 550.120

First, i just change the usd_path with my robot, and follow the `Modifying an existing Direct RL Environmentother` tutorial to modify the other parameters

No error but the app is stuck
```
`[INFO] Using python from: /home/lenovo/anaconda3/envs/isaaclab_binary/bin/python                                                                                                                                                                                
[INFO][AppLauncher]: Loading experience file: /home/lenovo/.local/share/ov/pkg/isaac-sim-4.2.0/IsaacLab/source/apps/isaaclab.python.kit
Loading user config located at: '/home/lenovo/.local/share/ov/pkg/isaac-sim-4.2.0/kit/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: /home/lenovo/.local/share/ov/pkg/isaac-sim-4.2.0/kit/logs/Kit/Isaac-Sim/4.2/kit_20250225_110912.log
2025-02-25 03:09:12 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.

|---------------------------------------------------------------------------------------------|
| Driver Version: 550.120       | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090          | Yes: 0 |     | 24564   MB | 10de      | 0          |
|     |                                  |        |     |            | 2684      | 23295802.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 22.04.4 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.4, Kernel: 6.5.0-41-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: 13th Gen Intel(R) Core(TM) i9-13900K | Cores: 24 | Logical: 32
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31711 | Free Memory: 25486
| Total Page/Swap (MB): 17903 | Free Page/Swap: 17903
|---------------------------------------------------------------------------------------------|
2025-02-25 03:09:19 [7,510ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
[INFO]: Parsing configuration from: omni.isaac.lab_tasks.direct.D12_cabinet.D12_cabinet_env:D12CabinetEnvCfg
[INFO]: Parsing configuration from: /home/lenovo/.local/share/ov/pkg/isaac-sim-4.2.0/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/D12_cabinet/agents/rl_games_ppo_cfg.yaml
[INFO] Logging experiment in directory: /home/lenovo/.local/share/ov/pkg/isaac-sim-4.2.0/IsaacLab/logs/rl_games/franka_cabinet_direct
Setting seed: 42
[2025-02-25 11:09:26,149][ogn_registration][INFO] - Looking for Python nodes to register in omni.physx.fabric-106.1.9
[2025-02-25 11:09:26,149][ogn_registration][INFO] -  -> Registered nodes from module omni.physxfabric at /home/lenovo/.local/share/ov/pkg/isaac-sim-4.2.0/extsPhysics/omni.physx.fabric
[2025-02-25 11:09:26,149][ogn_registration][INFO] - Registering nodes in /home/lenovo/.local/share/ov/pkg/isaac-sim-4.2.0/extsPhysics/omni.physx.fabric imported as omni.physxfabric with AutoNode config {}
[2025-02-25 11:09:26,149][ogn_registration][INFO] - Registering Python Node Types from omni.physxfabric at /home/lenovo/.local/share/ov/pkg/isaac-sim-4.2.0/extsPhysics/omni.physx.fabric in omni.physx.fabric
[2025-02-25 11:09:26,149][ogn_registration][INFO] - ========================================================================================================================
[2025-02-25 11:09:26,150][ogn_registration][INFO] - No dependency on omni.graph, therefore no nodes to register in omni.physx.fabric
[2025-02-25 11:09:26,150][ogn_registration][INFO] - ...None found, no registration to do
[2025-02-25 11:09:26,151][ogn_registration][INFO] - ...Skipping: No OmniGraph presence in the module omni.physxfabric - No nodes in this module, do not remember it
[2025-02-25 11:09:26,151][ogn_registration][INFO] - Destroying registration record for omni.physx.fabric
[2025-02-25 11:09:26,151][ogn_registration][INFO] - OGN register omni.physx.fabric-106.1.9 took 1468222.000000
[INFO]: Base environment:
    Environment device    : cuda:0
    Environment seed      : 42
    Physics step-size     : 0.008333333333333333
    Rendering step-size   : 0.016666666666666666
    Environment step-size : 0.016666666666666666
[INFO]: Time taken for scene creation : 1.575817 seconds
[INFO]: Scene manager:  <class InteractiveScene>
    Number of environments: 1024
    Environment spacing   : 3.0
    Source prim name      : /World/envs/env_0
    Global prim paths     : ['/World/ground']
    Replicate physics     : True
[INFO]: Starting the simulation. This may take a few seconds. Please wait...`

```

[... 85 lines cut ...]
    articulation.reset(env_ids)
  File "/home/lenovo/.local/share/ov/pkg/isaac-sim-4.2.0/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/articulation.py", line 170, in reset
    for actuator in self.actuators.values():
AttributeError: 'Articulation' object has no attribute 'actuators'

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
2025-02-25 03:39:11 [12,903ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-25 03:39:11 [12,904ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-25 03:39:11 [12,904ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-25 03:39:11 [12,904ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-25 03:39:11 [12,904ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-25 03:39:11 [12,904ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-25 03:39:11 [12,904ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-25 03:39:11 [12,904ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-25 03:39:11 [12,904ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-25 03:39:11 [12,904ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-25 03:39:11 [12,904ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-25 03:39:11 [12,904ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-02-25 03:39:11 [12,904ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2025-02-25 03:39:11 [13,059ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
```

The app showed the robot but it closed instantly due to the error.

Could someone gives some help?

RandomOakForest, 2025-02-27:
Thank you for posting this. Would it be possible for you to try Isaac Sim 4.5 and Isaac Lab 2.0? then could you try just loading the robot, as in using [the steps in this tutorial](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/01_assets/run_articulation.html)? Thanks.

closed 2025-03-21 by RandomOakForest

Ozzey, 2025-03-26:
Your join t_pos values for the robot need to be float. For example:

 ```
init_state=ArticulationCfg.InitialStateCfg(
            joint_pos={
                "joint1": 0.0,
                "joint2": 0.0,
                "joint3": 0.0,
                "joint4": 0.0,
                "joint5": 0.0,
                "joint6": 0.0,
                "joint7": 0.0,
                "joint8": 0.0,
            },
            pos=(0.8, 0.0, 0.5),
            rot=(0.0, 0.0, 0.0, 1.0),
        ),
```