isaac-sim/IsaacLab #1936: [Bug Report] Upgrade from IsaacSim 4.2 to IsaacSim 4.5 + isaacLab, old scirpt with pinocchio has libhpp-fcl.so: undefined symbol

cidxb, 2025-02-25:
Problem:
I have a script that includes using pinoccio==2.7.0 to do inverse kinematics. 
Now I am testing to migrate it to isaacsim==4.5+Isaaclab 2.0. But it reported back 
```

Traceback (most recent call last):
  File "/home/liangyi/workspace/px_LearningSim_Janus/example/IsaacSim_run_ik_interactivescene_tora.py", line 68, in <module>
    from px_janus_learnsim.robot.controllers.pinocchio_test import Px_kinematics
  File "/home/liangyi/workspace/px_LearningSim_Janus/src/px_janus_learnsim/robot/controllers/pinocchio_test.py", line 9, in <module>
    import pinocchio as pin
  File "/home/liangyi/anaconda3/envs/test/lib/python3.10/site-packages/cmeel.prefix/lib/python3.10/site-packages/pinocchio/__init__.py", line 18, in <module>
    from .pinocchio_pywrap import *
ImportError: /home/xxx/anaconda3/envs/test/lib/python3.10/site-packages/cmeel.prefix/lib/python3.10/site-packages/pinocchio/../../../../lib/libhpp-fcl.so: undefined symbol: _ZNK6Assimp8IOSystem16CurrentDirectoryB5cxx11Ev
```

Normally we should go after to check the library itself but after testing. The same version library works well in Isaac sim 4.2. with the old script.

So that's why I raise this up in the community.
If anyone knows how to duel with it, much grateful.
Thanx

 System : Ubuntu 22.04

cidxb, 2025-02-25:
I have checked https://github.com/isaac-sim/IsaacLab/issues/271 seems to be the same problem but no solution

cidxb, 2025-02-25:
Well, I found a solution :
change the import order; I have encountered similar issues before, while the import order is heavily important for the normal running of the script. 

Still it reported back problem , but now the script is running:
```
2025-02-25 09:59:22 [2,823ms] [Error] [omni.ext._impl.custom_importer] Failed to import python module isaacsim.robot_motion.lula. Error: /home/xxx/anaconda3/envs/test/lib/python3.10/site-packages/isaacsim/exts/isaacsim.robot_motion.lula/pip_prebundle/_lula_libs/liblula_kinematics.so: undefined symbol: _ZN4urdf9parseURDFERKSs. Traceback:
Traceback (most recent call last):
  File "/home/xx/anaconda3/envs/test/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/custom_importer.py", line 76, in import_module
    return importlib.import_module(name)
  File "/home/xxx/anaconda3/envs/test/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/xxx/anaconda3/envs/test/lib/python3.10/site-packages/isaacsim/exts/isaacsim.robot_motion.lula/isaacsim/robot_motion/lula/__init__.py", line 10, in <module>
    from .extension import *
  File "/home/xxx/anaconda3/envs/test/lib/python3.10/site-packages/isaacsim/exts/isaacsim.robot_motion.lula/isaacsim/robot_motion/lula/extension.py", line 24, in <module>
    from lula import LogLevel, set_default_logger_prefix, set_log_level
ImportError: /home/xxx/anaconda3/envs/test/lib/python3.10/site-packages/isaacsim/exts/isaacsim.robot_motion.lula/pip_prebundle/_lula_libs/liblula_kinematics.so: undefined symbol: _ZN4urdf9parseURDFERKSs

2025-02-25 09:59:22 [2,823ms] [Error] [carb.scripting-python.plugin] Exception: Extension python module: 'isaacsim.robot_motion.lula' in '/home/xxx/anaconda3/envs/test/lib/python3.10/site-packages/isaacsim/exts/isaacsim.robot_motion.lula' failed to load.

At:
  /home/xxx/anaconda3/envs/test/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/_internal.py(213): startup
  /home/xxx/anaconda3/envs/test/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/_internal.py(328): startup_extension
  PythonExtension.cpp::startup()(2): <module>
  /home/xxx/anaconda3/envs/test/lib/python3.10/site-packages/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py(427): _start_app
  /home/xxx/anaconda3/envs/test/lib/python3.10/site-packages/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py(212): __init__
  /home/xxx/workspace/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py(638): _create_app
  /home/xxx/workspace/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py(114): __init__
  /home/xxx/workspace/px_LearningSim_Janus/example/IsaacSim_run_ik_interactivescene_tora.py(41): <module>

2025-02-25 09:59:22 [2,823ms] [Error] [omni.ext.plugin] [ext: isaacsim.robot_motion.lula-4.0.4] Failed to startup python extension.
[3.117s] [ext: omni.usd.metrics.assembler.ui-106.5.0] startup
[3.120s] [ext: isaacsim.asset.browser-1.3.4] startup
[3.163s] [ext: isaacsim.gui.menu-2.0.9] startup
2025-02-25 09:59:22 [3,149ms] [Warning] [omni.kit.menu.utils.app_menu] add_menu_items: menu [<MenuItemDescription name:'New'>, <MenuItemDescription name:'Open'>, <MenuItemDescription name:'Re-open with New Edit Layer'>, <MenuItemDescription name:'Save'>, <MenuItemDescription name:'Save With Options'>, <MenuItemDescription name:'Save As...'>, <MenuItemDescription name:'Save Flattened As...'>, <MenuItemDescription name:'Add Reference'>, <MenuItemDescription name:'Add Payload'>, <MenuItemDescription name:'Exit'>] cannot change delegate
[3.167s] [ext: omni.kit.ui.actions-1.0.2] startup
[3.170s] [ext: omni.kit.window.console-0.2.14] startup
[3.178s] [ext: isaacsim.robot_motion.motion_generation-8.0.6] startup
2025-02-25 09:59:22 [3,168ms] [Error] [omni.ext._impl.custom_importer] Failed to import python module isaacsim.robot_motion.motion_generation. Error: /home/xxx/anaconda3/envs/test/lib/python3.10/site-packages/isaacsim/exts/isaacsim.robot_motion.lula/pip_prebundle/_lula_libs/liblula_kinematics.so: undefined symbol: _ZN4urdf9parseURDFERKSs. Traceback:
Traceback (most recent call last):
  File "/home/xxx/anaconda3/envs/test/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/custom_importer.py", line 76, in import_module
    return importlib.import_module(name)
  File "/home/xxx/anaconda3/envs/test/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
[... cut ...]
  /home/xxx/workspace/px_LearningSim_Janus/example/IsaacSim_run_ik_interactivescene_tora.py(41): <module>

2025-02-25 09:59:22 [3,168ms] [Error] [omni.ext.plugin] [ext: isaacsim.robot_motion.motion_generation-8.0.6] Failed to startup python extension.
[3.328s] [ext: isaaclab-0.34.1] startup
2025-02-25 09:59:22 [3,319ms] [Error] [omni.graph.core._impl.extension] OGN node registration completed with errors: /home/xxx/anaconda3/envs/test/lib/python3.10/site-packages/isaacsim/exts/isaacsim.robot_motion.lula/pip_prebundle/_lula_libs/liblula_kinematics.so: undefined symbol: _ZN4urdf9parseURDFERKSs

```

kellyguo11, 2025-02-25:
We've been looking into using pinocchio internally as well and found that using a conda environment with pinocchio installed through conda helps with the compatibility issues. This could be a workaround while we work on solving this in kit

cidxb, 2025-02-26:
> We've been looking into using pinocchio internally as well and found that using a conda environment with pinocchio installed through conda helps with the compatibility issues. This could be a workaround while we work on solving this in kit

Thank you for sharing the update

closed 2025-03-21 by RandomOakForest