isaac-sim/IsaacLab #1937: [Bug Report] Error Verifying Isaac Lab installation: ModuleNotFoundError: No module named 'omni.kit.usd', Failed to resolve extension dependencies, Synced registries

NoahSTEMStuff, 2025-02-25:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug
I installed isaac sim via binaries, and isaac sim is able to run fine, however, when trying to run the python script under the 
"Verifying the Isaac Lab installation" section I get an error with kit.

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

```
./isaaclab.sh -p scripts/tutorials/00_sim/create_empty.py
[INFO] Using python from: /home/noah/IsaacLab/_isaac_sim/python.sh                                                                        
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: /home/noah/IsaacLab/apps/isaaclab.python.kit
Loading user config located at: '/home/noah/isaacsim/kit/data/Kit/Isaac-Sim/4.5/user.config.json'
[Info] [carb] Logging to file: /home/noah/isaacsim/kit/logs/Kit/Isaac-Sim/4.5/kit_20250225_010701.log
2025-02-25 09:07:01 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.043s] Failed to solve some dependencies locally, syncing with extension registry...
[0.046s] [ext: omni.kit.async_engine-0.0.1] startup
[0.050s] [ext: omni.client.lib-1.0.0] startup
[0.071s] [ext: omni.kit.registry.nucleus-0.0.0] startup
2025-02-25 09:07:02 [68ms] [Error] [omni.kit.registry.nucleus.index_v2.index_sync] Syncing registry index failed for url: 'https://ovextensionsprod.blob.core.windows.net/exts/kit/prod/106/shared/v2'. Exception: Traceback (most recent call last):
  File "/home/noah/isaacsim/kit/extscore/omni.kit.registry.nucleus/omni/kit/registry/nucleus/index_v2/index_sync.py", line 116, in _task
    self._task_uncaught()
  File "/home/noah/isaacsim/kit/extscore/omni.kit.registry.nucleus/omni/kit/registry/nucleus/index_v2/index_sync.py", line 136, in _task_uncaught
    self._registry_file = RegistryFile(self._local_path)
  File "/home/noah/isaacsim/kit/extscore/omni.kit.registry.nucleus/omni/kit/registry/nucleus/index_v2/registry_file.py", line 28, in __init__
    data = json.loads(f.read())
  File "/home/noah/IsaacLab/_isaac_sim/kit/python/lib/python3.10/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/home/noah/IsaacLab/_isaac_sim/kit/python/lib/python3.10/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/home/noah/IsaacLab/_isaac_sim/kit/python/lib/python3.10/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

2025-02-25 09:07:02 [69ms] [Error] [omni.kit.registry.nucleus.index_v2.index_sync] Syncing registry index failed for url: 'https://ovextensionsprod.blob.core.windows.net/exts/kit/prod/sdk/106.5/d02c707b/v2'. Exception: Traceback (most recent call last):
  File "/home/noah/isaacsim/kit/extscore/omni.kit.registry.nucleus/omni/kit/registry/nucleus/index_v2/index_sync.py", line 116, in _task
    self._task_uncaught()
  File "/home/noah/isaacsim/kit/extscore/omni.kit.registry.nucleus/omni/kit/registry/nucleus/index_v2/index_sync.py", line 136, in _task_uncaught
    self._registry_file = RegistryFile(self._local_path)
  File "/home/noah/isaacsim/kit/extscore/omni.kit.registry.nucleus/omni/kit/registry/nucleus/index_v2/registry_file.py", line 28, in __init__
    data = json.loads(f.read())
  File "/home/noah/IsaacLab/_isaac_sim/kit/python/lib/python3.10/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/home/noah/IsaacLab/_isaac_sim/kit/python/lib/python3.10/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/home/noah/IsaacLab/_isaac_sim/kit/python/lib/python3.10/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

2025-02-25 09:07:02 [489ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
	[isaaclab.python-2.0.1] dependency: 'omni.kit.menu.edit' = { version='^' } can't be satisfied. Available versions:
	 (none found)
 Synced registries:
	 - kit/default         : found 0 packages (couldn't connect or empty)
	 - kit/sdk             : found 0 packages (couldn't connect or empty)
	 - kit/community       : found 399 packages 

2025-02-25 09:07:02 [490ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...
Traceback (most recent call last):
  File "/home/noah/IsaacLab/scripts/tutorials/00_sim/create_empty.py", line 29, in <module>
    app_launcher = AppLauncher(args_cli)
  File "/home/noah/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py", line 114, in __init__
    self._create_app()
  File "/home/noah/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py", line 638, in _create_app
    self._app = SimulationApp(self._sim_app_config, experience=self._sim_experience_file)
  File "/home/noah/IsaacLab/_isaac_sim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py", line 227, in __init__
    from .utils import create_new_stage, open_stage
[... 184 lines cut ...]
  File "/home/noah/Downloads/isaac-sim-comp-check@4.5.0-rc.6+release.675.f1cca148.gl.linux-x86_64.release/kit/python/lib/python3.10/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

2025-02-25 08:59:43 [2,034ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
	[omni.app.mini-0.1.1] dependency: 'omni.kit.uiapp' = { version='^' } can't be satisfied. Available versions:
	 (none found)
 Synced registries:
	 - kit/default         : found 0 packages (couldn't connect or empty)
	 - kit/sdk             : found 0 packages (couldn't connect or empty)
	 - kit/prod/default    : found 0 packages (couldn't connect or empty)
	 - kit/prod/sdk        : found 0 packages (couldn't connect or empty)

2025-02-25 08:59:43 [2,035ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...
```

### Checklist

- [ ] I have checked that there is no similar issue in the repo - there are some similar ones, but there was either no solution, the post was closed, and it was not applicable for my setup
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo - although I get a similar error when running the Isaac sim compatibility error, Isaac sim the errors appear, but it doesn't seem to affect isaac sim performance, only isaac lab

### Acceptance Criteria

- [ ] Criteria 1: ```./isaaclab.sh -p scripts/tutorials/00_sim/create_empty.py``` runs with no issue
- [ ] Criteria 2: I am able to finish the installing Isaac lab tutoring and train a robot

RandomOakForest, 2025-02-27:
Thank you for posting this. Could you let us know which steps you followed to install Isaac Lab? Thanks.

NoahSTEMStuff, 2025-02-28:
Yeah, first I installed isaac sim through binaries(https://docs.isaacsim.omniverse.nvidia.com/latest/installation/install_workstation.html) then did 
```
git clone git@github.com:isaac-sim/IsaacLab.git
# enter the cloned repository
cd IsaacLab
# create a symbolic link
ln -s ${HOME}/isaacsim _isaac_sim
# I skipped 'sudo apt install cmake build-essential' since they were already installed
./isaaclab.sh --install
```

But I believe I fixed my issue, I ran the following commands:
```
mv ~/.nvidia-omniverse ~/.nvidia-omniverse.bak
mv ~/.cache/ov ~/.cache/ov.bak
mv ~/.local/share/ov ~/.local/share/ov.bak
```

Now I can run the Isaac sim compatibility checker, Isaac Sim, and Isaac lab without crashing and without any errors involving synced registries. Training a policy also works fine.

closed 2025-02-28 by NoahSTEMStuff

Pipotheavenger, 2025-06-26:
Thanks Bro @MiangChen you saved all my project ! Thanks for posting the solution