isaac-sim/IsaacLab #1942: [Question] Cpu memory usage keep increasing when using "gym.wrappers.RecordVideo"

AndreaRossetto, 2025-02-25:
### Question

I'm trying to train a PPO (skrl) and record the training, this is the code I wrote to train and save the video of the enviroment

```
    # create isaac environment
    env = RobotEnv(cfg=env_cfg, render_mode="rgb_array", headless=args_cli.headless)

    # convert to single-agent instance if required by the RL algorithm
    if isinstance(env.unwrapped, DirectMARLEnv) and algorithm in ["ppo"]:
        env = multi_agent_to_single_agent(env)

    # adjust camera resolution and pose
    env_cfg.viewer.resolution = (1920, 1080)
    env_cfg.viewer.eye = (1.0, 1.0, 1.0)
    env_cfg.viewer.lookat = (0.0, 0.0, 0.0)

    # wrap for video recording
    if args_cli.video == True:
        video_kwargs = {
            "video_folder": os.path.join(log_dir, "videos", "train"),
            "step_trigger": lambda step: step % args_cli.video_interval == 0,
            "video_length": args_cli.video_length,
            "disable_logger": True,
        }
        print("[INFO] Recording videos during training.")
        print_dict(video_kwargs, nesting=4)
        env = gym.wrappers.RecordVideo(env, **video_kwargs)

    # wrap around environment for skrl
    env = SkrlVecEnvWrapper(env, ml_framework=args_cli.ml_framework)  # same as: `wrap_env(env, wrapper="auto")`

```
I'm facing some problems of memory allocation, during time the memory usage keep increasing until it saturates and the process get killed by Ubuntu. if I remove this part the memory usage still constant during the training.

```
    if args_cli.video == True:
        video_kwargs = {
            "video_folder": os.path.join(log_dir, "videos", "train"),
            "step_trigger": lambda step: step % args_cli.video_interval == 0,
            "video_length": args_cli.video_length,
            "disable_logger": True,
        }
        print("[INFO] Recording videos during training.")
        print_dict(video_kwargs, nesting=4)
        env = gym.wrappers.RecordVideo(env, **video_kwargs)
```

Why the memory keep increasing? What am I doing wrong?

RandomOakForest, 2025-03-01:
Thank you for posting this question. It is hard to say from the code you post what could be missing. Could you please file a _Bug Report_ with steps for us to reproduce the issue, if you still face this issue? Thanks.

AndreaRossetto, 2025-03-02:
Yes, I'm still facing the problem at the moment I'm not using that to avoid the issue. I'll open a Bug Report to reproduce the issue.

YannickP99, 2025-03-04:
I got the same problem using the new IsaacLab version. Without video recording the memory stays at ~10GB, but when using the recording the memory slowly increases until the simulation breaks at about 500 episodes (32GB memory)

AndreaRossetto, 2025-03-04:
I've opened a [bug report](https://github.com/isaac-sim/IsaacLab/issues/1996)

closed 2025-03-21 by RandomOakForest