isaac-sim/IsaacLab #1951: [Bug Report] Isaac Lab with Isaac Sim pip Installation Failed to setup VSCode

j3soon, 2025-02-26:
### Describe the bug

Cannot setup VSCode when using Isaac Lab with Isaac Sim pip Installation.

Issue https://github.com/isaac-sim/IsaacLab/issues/629 has a related discussion on this. However, I'm unsure whether this issue was never fully resolved or if it is a regression in Isaac Lab v2.x. Another user also encountered this issue and commented in <https://github.com/isaac-sim/IsaacLab/issues/629#issuecomment-2408850402>.

### Steps to reproduce

1. Install Isaac Lab with Isaac Sim pip (<https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html>)
2. Attempt to setup VSCode (<https://isaac-sim.github.io/IsaacLab/main/source/overview/developer-guide/vs_code.html>) and execute `Ctrl+Shift+P > Tasks: Run Task > setup_python_env`, observe the following error:

```
 *  Executing task: /home/johnson/IsaacLab/isaaclab.sh -p /home/johnson/IsaacLab/.vscode/tools/setup_vscode.py 

[INFO] Using python from: /home/johnson/miniconda3/envs/env_isaaclab/bin/python                                                             
[WARN] Could not find Isaac Sim VSCode settings: /home/johnson/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/isaacsim/.vscode/settings.json.
    This will result in missing 'python.analysis.extraPaths' in the VSCode
    settings, which limits the functionality of the Python language server.
    However, it does not affect the functionality of the Isaac Lab project.
    We are working on a fix for this issue with the Isaac Sim team.
 *  Terminal will be reused by tasks, press any key to close it. 
```

### System Info

Describe the characteristic of your environment:

- Commit: f1a4975eb7bae8509082a8ff02fd775810a73531
- Isaac Sim Version: 4.5.0 (pip)
- OS: Ubuntu 22.04.5 LTS
- GPU: L40
- CUDA: 11.8
- GPU Driver: 550.127.05

### Additional context

Further investigation:

```
$ ls -a /home/johnson/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/isaacsim/
.   apps  extscache       extsPhysics  LICENSE.txt  VERSION
..  exts  extsDeprecated  __init__.py  __pycache__
```

It can be observed that Isaac Sim 4.5.0 pip package doesn't include the `.vscode` folder.

PR https://github.com/isaac-sim/IsaacLab/pull/631 is a partial fix to this issue, it doesn't solve this issue since the Isaac Sim upstream package doesn't provide the `.vscode/settings.json` file.

This issue also exists on Windows 11: <https://github.com/isaac-sim/IsaacLab/issues/629#issuecomment-2408850402>.

There may be 2 ways to resolve this:
1. Ask upstream to include the `.vscode` directory in all future Isaac Sim pip packages.
2. Include the full `.vscode/settings.json` file directly in the Isaac Lab repo, removing the need to depend on upstream Isaac Sim settings.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

kellyguo11, 2025-02-26:
The VSCode setup is only available for binary installed Isaac Sim currently. We will update the docs to note this.

j3soon, 2025-02-27:
Got it, thanks! In addition, should we unify the Isaac Sim binary installation path?

- In Isaac Sim 4.5.0, the suggested installation path is `~/isaacsim` on Linux, `C:\isaacsim` on Windows. (https://docs.isaacsim.omniverse.nvidia.com/latest/installation/install_workstation.html)
- In Isaac Lab 2.0.1, the suggested installation path is `${HOME}/Downloads/isaac-sim-standalone@4.5.0` on Linux and `%USERPROFILE%/Downloads/isaac-sim-standalone@4.5.0` on Windows. (https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/binaries_installation.html)

I wonder if this choice is intended? If not, I think it may be better to follow Isaac Sim's installation location (`~/isaacsim` and `C:\isaacsim`) for simplicity.

If this is preferred, I can open a PR to update the documentation on these two cases (VSCode setup only supports binary installed Isaac Sim & Unify Isaac Sim installation path).

As a side note, the default binary filename is `isaac-sim-standalone@4.5.0-rc.36+release.19112.f59b3005.gl.linux-x86_64.release.zip` for Linux and `isaac-sim-standalone@4.5.0-rc.36+release.19112.f59b3005.gl.windows-x86_64.release.zip` for Windows, so we're going to rename it anyway.

kellyguo11, 2025-02-27:
Sounds good to me, we would welcome the PR.

j3soon, 2025-03-01:
Related: https://github.com/isaac-sim/IsaacLab/discussions/1983

closed 2025-03-03 by kellyguo11