isaac-sim/IsaacLab #1952: [Bug Report] terrain.env_origins bug in DirectRLEnv

xliu0105, 2025-02-26:
### Describe the bug

When I create my own DirectRLEnv environment, I set `default_root_state[:, :3] += self._terrain.env_origins[env_ids]` in the `_reset_idx` function, and the following series of errors are reported:
```
../aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [0,0,0], thread: [3,0,0] Assertion -sizes[i] <= index && index < sizes[i] && "index out of bounds" failed.

RuntimeError: CUDA error: device-side assert triggered CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect. For debugging consider passing CUDA_LAUNCH_BLOCKING=1 Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.
```
But when I use `default_root_state[:, :3] += self.scene.env_origins[env_ids]`, there will be no such error.

In `Isaaclab/source/isaaclab_tasks/isaaclab_tasks/direct/anymal_c/anymal_c_env.py`, `default_root_state[:, :3] += self._terrain.env_origins[env_ids]` is used, but this will not report an error. I carefully checked my code and don't know where the problem lies.

### Steps to reproduce

None

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: Isaac Sim 4.5.0
- OS: Ubuntu22.04
- GPU: RTX 4060 Ti 16G
- CUDA: 12.1
- GPU Driver: 

### Additional context

Add any other context about the problem here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-03-01:
Thanks for posting this. If you could add an example on "steps to reproduce" would be helpful. We will investigate.

Toni-SM, 2025-03-01:
Hi @xliu0105 

Make sure the `default_root_state` is indexed using the given `env_ids` indexes

```
default_root_state = self._robot.data.default_root_state[env_ids]
default_root_state[:, :3] += self._terrain.env_origins[env_ids]
```

or use the `env_ids` indexes explicitly if the `default_root_state` was not indexed before

```
default_root_state = self._robot.data.default_root_state
default_root_state[env_ids, :3] += self._terrain.env_origins[env_ids]
```

Mayankm96, 2025-03-02:
Can you tell us the PyTorch version you are using? We saw a similar issue on 5090 series and had to upgrade torch to use CUDA 12.8.

```
pip install --pre torch --index-url https://download.pytorch.org/whl/nightly/cu128 --force-reinstall
```

xliu0105, 2025-03-03:
> Can you tell us the PyTorch version you are using? We saw a similar issue on 5090 series and had to upgrade torch to use CUDA 12.8.
> 
> ```
> pip install --pre torch --index-url https://download.pytorch.org/whl/nightly/cu128 --force-reinstall
> ```
My system:
Ubuntu-22.04
GPU: 4060Ti 16G
Pytorch: 2.5.1
CUDA: 12.1

closed 2025-07-07 by kellyguo11