isaac-sim/IsaacLab #1957: [Bug Report] Reshape + shape computation issues with observation manager

pculbertson-bdai, 2025-02-27:
#### Description
When concatenating observation terms with history enabled, the Observation Manager computes the combined shape incorrectly by summing the history dimension for each term (e.g. turning [(H, D1), (H, D2), (H, D3)] into (3*H, D1+D2+D3) instead of (H, D1+D2+D3)). 

Additionally, observation terms with arbitrary trailing dimensions (beyond a single feature dimension) were not handled correctly. This PR fixes these issues by:
- Preserving the common history dimension when concatenating terms.
- Supporting unflattened terms with arbitrary trailing dimensions by summing only the last dimension while ensuring the preceding trailing dimensions match.

#### Steps to Reproduce
1. Create an observation group with multiple history-enabled terms. For example:
```
@configclass
class ObservationManagerCfg:
    @configclass
    class PolicyCfg(ObservationGroupCfg):
        history_length = H
        flatten_history_dim = False
        term1 = ObservationTermCfg(func=f1) # f1 returns shape (D1,)
        term2 = ObservationTermCfg(func=f2) # f2 returns shape (D2,)

    policy = PolicyCfg()
```
*Expected:*  Upon instantiating the env, `env.observation_space["policy"].shape` is  `(num_envs, H, D1 + D2)`. 
*Actual*: Upon instantiating the env, `env.observation_space["policy"].shape` is  `(num_envs, 2H, D1 + D2)`. 

2. Create an observation group with `flatten_history_dim=True` (e.g., change this parameter of the last example.

When reading an observation from the observation manager,
```
policy_obs = observation_manager.compute()["policy"]
```
*Expected:* `policy_obs.reshape(-1,  H, D1 + D2) = torch.concatenate([term1, term2], dim=-1)`. Note this is poor syntax - but ideally we could reconstruct the observations with a coherent history dim.
*Actual:* `policy_obs = torch.concatenate([term1.reshape(num_envs, -1), term2.reshape(num_envs, -1)], dim=-1).` The reshape does not preserve a history dimension.

### System Info

Describe the characteristic of your environment:
- Commit: f1a4975eb7bae8509082a8ff02fd775810a73531
- Isaac Sim Version: 4.5.0
- OS: Ubuntu 22.04
- GPU: RTX 5000 Ada
- CUDA: 12.2
- GPU Driver: 535.183.01

### Additional context

Will open a PR with suggested changes to fix these issues.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] Concatenated, unflattened histories have the correct shape in `env.observation_space`
- [ ] History observations can be reshaped easily to preserve a history dim.

RandomOakForest, 2025-03-01:
Thanks for posting this. The team will review.

jtigue-bdai, 2025-04-02:
This may also be addressing the further concerns of #1208 of flattened history order ABCABC instead of AABBCC

StafaH, 2026-09-27:
This is now fully fixed by #8079 (the shape bug was fixed in a previous PR)

closed 2026-09-27 by StafaH