isaac-sim/IsaacLab #1969: [Proposal] Add Support for Virtual Xform Creation in ManagerBasedEnv

right-chan, 2025-02-28:
### Proposal

This proposal suggests adding support for creating virtual `Xform` entities in `ManagerBasedEnv` configurations within IsaacLab. The primary goal is to provide a clean and flexible way to define additional coordinate frames without modifying existing robot assets directly. This feature would enable users to dynamically add transformations (such as virtual camera frames) as part of their environment configuration.

### Motivation

When designing ManagerBasedEnv, users often rely on pre-existing assets. However, there are cases where additional geometry or transformations need to be introduced dynamically. One such case is adding a virtual coordinate frame to handle specific transformations, such as avoiding gimbal lock or aligning sensors correctly.

For example, when attempting to spawn a camera at `prim_path="{ENV_REGEX_NS}/Robot/panda_hand/camera_frame/camera"`, the asset structure only includes `panda_hand`, making it impossible to create a child entity at `camera_frame`. The only alternative is modifying the robot asset itself, which is not always desirable.

By introducing an `XformCfg` configuration class, users can define and spawn an `Xform` at the desired location, allowing the camera (or other entities) to be correctly placed without modifying the original asset.

### Alternatives

Modifying the Robot Asset: Users could manually update the asset(usd) definition to include the additional frames, but this is cumbersome and not scalable across different environments.

### Proposed Implementation

The following implementation introduces an `XformCfg` class to enable dynamic `Xform` creation:
```python
from dataclasses import MISSING
from typing import Callable
from isaaclab.utils import configclass
from isaaclab.sim import SpawnerCfg
from isaaclab.sim import clone
from isaaclab.assets import AssetBaseCfg
import isaacsim.core.utils.prims as prim_utils
import omni.log

@clone
def spawn_geometry(prim_path, cfg, translation, orientation):
    # spawn asset if it doesn't exist.
    if not prim_utils.is_prim_path_valid(prim_path):
        # add prim as reference to stage
        prim_utils.create_prim(
            prim_path,
            prim_type="Xform",
            translation=translation,
            orientation=orientation,
        )
    else:
        omni.log.warn(f"A prim already exists at prim path: '{prim_path}'.")
    # return the prim
    return prim_utils.get_prim_at_path(prim_path)

@configclass
class GeometryCfg(SpawnerCfg):
    func: Callable = spawn_geometry

@configclass
class XformCfg(AssetBaseCfg):
    prim_path: str = MISSING
    spawn: GeometryCfg = GeometryCfg()
    init_state = AssetBaseCfg.InitialStateCfg()
```

### Example Usage
This implementation allows users to define XformCfg in their scene configuration, making it easier to manage additional transformations without modifying assets.
```python
from isaaclab.scene import InteractiveScene, InteractiveSceneCfg
from isaaclab_assets.robots.franka import FRANKA_PANDA_CFG
import isaaclab.sim as sim_utils
from isaaclab.sensors import CameraCfg

import XformCfg

@configclass
class MySceneCfg(InteractiveSceneCfg):
    robot = FRANKA_PANDA_CFG

    camera_frame = XformCfg(
        prim_path="{ENV_REGEX_NS}/Robot/panda_hand/camera_frame",
        init_state=XformCfg.InitialStateCfg(rot=(0.0, 0.70711, 0.70711, 0.0)),
    )

    camera = CameraCfg(
        prim_path="{ENV_REGEX_NS}/Robot/panda_hand/camera_frame/camera",
        spawn=sim_utils.PinholeCameraCfg(),
        offset=CameraCfg.OffsetCfg(pos=(0.025, 0.0, 0.0), rot=(0.7071, 0.0, 0.0, 0.7071), convention="ros"),
    )

scene = InteractiveScene(MySceneCfg())
```

### Additional Context
This feature would improve the usability of `ManagerBasedEnv` by allowing dynamic frame creation without requiring modifications to the core asset definitions. It ensures a modular and reusable approach for defining transformations in complex robotic environments.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

RandomOakForest, 2025-03-01:
Thank you for posting this proposal. The team will review.

Mayankm96, 2025-03-01:
Is the ask to add a spawner method to make an empty Xform?

right-chan, 2025-03-02:
Yes, the request is to add a spawner method that dynamically creates an empty Xform within `ManagerBasedEnv` configurations.

The motivation is to enable users to define additional coordinate frames (e.g., for camera alignment) without modifying existing robot assets directly.

AntoineRichard, 2026-08-27:
@right-chan thanks for the well-scoped proposal — a physics-free child frame for mounting sensors is a real gap, and it's now filled.

`isaaclab.sim.spawners.sensors.SensorFrameCfg` spawns a plain USD `Xform` with no rigid-body or collision API, usable directly as a scene entry:

```python
frame = AssetBaseCfg(
    prim_path="{ENV_REGEX_NS}/Robot/panda_hand/cam_frame",
    spawn=sim_utils.SensorFrameCfg(),
    init_state=AssetBaseCfg.InitialStateCfg(pos=(0.0, 0.0, 0.1), rot=(0.0, 0.0, 0.0, 1.0)),
)
```

It works on all backends including Newton. Note Isaac Lab 3.0 uses `(x, y, z, w)` ordering, so identity is `(0, 0, 0, 1)`. Closing as implemented — reopen if it misses your camera-frame case.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard