isaac-sim/IsaacLab #1971: [Question] How can I make an object in my scene move according to a given trajectory.

HilbertXu, 2025-02-28:
### Question

Hi, I'm currently working on a pick and place env with a panda arm and an object. Now, I want to make the object move with a pre-collected object pose sequence. The script I'm using is based on the [run_diff_ik.py ](https://github.com/isaac-sim/IsaacLab/blob/main/scripts/tutorials/05_controllers/run_diff_ik.py)example. I added a custom .usd object in the following way, 

````
obj = RigidObjectCfg(
        prim_path="{ENV_REGEX_NS}/Object",
        spawn=sim_utils.UsdFileCfg(
            usd_path='/path/to./usd', scale=(1.0, 1.0, 1.0),
            rigid_props=sim_utils.RigidBodyPropertiesCfg(disable_gravity=True),
            mass_props=sim_utils.MassPropertiesCfg(mass=0.05),
            collision_props=sim_utils.CollisionPropertiesCfg(collision_enabled=False),
            articulation_props=sim_utils.ArticulationRootPropertiesCfg(articulation_enabled=True, fix_root_link=False),
            visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 0.75, 0.0)),
        ),
        init_state=RigidObjectCfg.InitialStateCfg(),
    )
````
and then try to set its pose by calling the `write_root_pose_to_sim` function of the RigidObject istance I got from the scene `obj = scene["obj]`.  Then followed by calling the following lines to write the data to sim and step:

````
scene.write_data_to_sim()
# perform step
sim.step()
````

However, I found that the object remains still during the simulation. Could you please tell me how can I make the object move according to a given trajectory? 

ps: you can find the script I used below:
[run_diff_ik.txt](https://github.com/user-attachments/files/19032698/run_diff_ik.txt)

Many thanks in advance!

Yucheng

RandomOakForest, 2025-03-01:
Thanks for posting this. You will need to work with [waypoints](https://isaac-sim.github.io/IsaacLab/main/source/api/lab_mimic/isaaclab_mimic.datagen.html#waypoint-trajectory). Also it may be helpful, in the Isaac Sim docs those related to [motion generation, as in this example](https://docs.isaacsim.omniverse.nvidia.com/latest/manipulators/manipulators_lula_trajectory_generator.html#simple-case-linearly-connecting-waypoints.

HilbertXu, 2025-03-01:
Hi @RandomOakForest ,

Thank you very much for your intermediate response! I'm wondering if there are any examples/tutorials regarding using waypoints, especially about how to set waypoint for and RigidObject? I can only find one usage of waypoint in the isaaclab_mimic. I've also checked the motion generation in isaac sim docs, however, seems like the APIs in isaaclab are not as same as in the isaac sim.

Cheers,
Yucheng

RandomOakForest, 2025-04-03:
Thanks for following up. Unfortunately, the docs posted above are the current relevant docs we can provide. Regarding the follow up question, if you still face this issue, please submit a _Bug Report_ if you haven't. Thanks.

closed 2025-04-03 by RandomOakForest