isaac-sim/IsaacLab #1976: [Bug Report] [Question] VSCode Run and Debug error - no module named toml and incorrect launch.json

NNoah18, 2025-03-01:
### Describe the bug

I cannot run and debug using any of the configurations that are generated for VSCode. I also believe there is a mistake in some of the paths in the launch.json file in vscode.

### Steps to reproduce

I followed the steps to generate the .json files, however I am still unable to run and debug inside of VSCode. I ran into 2 issues
1.  the program path is incorrect. For some reason it generate paths with "standalone"  included which does not exist - snippet from the generated launch.json file. 
       ``` {
            "name": "Python: Train Environment",
            "type": "python",
            "request": "launch",
            "args" : ["--task", "Isaac-Reach-Franka-v0", "--headless"],
            "program": "${workspaceFolder}/source/standalone/workflows/rsl_rl/train.py",
            "console": "integratedTerminal"
        },```
this causes issues since ```${workspaceFolder}/source/standalone/workflows/rsl_rl/train.py"``` doesn't exist, so I manually changed it, but am unsure if this is a bug or the fact that I had to manually change it means I messed something up previously. I changed it too: 
``` "${workspaceFolder}/scripts/reinforcement_learning/rsl_rl/train.py"``` and now it is able to at least find the script.
2. After making the modification in 1 and  clicking run and debug for that configuration, I get the following error: 

```
$  cd /home/noah/IsaacLab ; /usr/bin/env /home/noah/IsaacLab/_isaac_sim/kit/python/bin/python3 /home/noah/.vscode/extensions/ms-python.debugpy-2025.0.1-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher 45371 -- /home/noah/IsaacLab/scripts/reinforcement_learning/rsl_rl/train.py --task Isaac-Reach-Franka-v0 --headless 
Traceback (most recent call last):
  File "/home/noah/IsaacLab/_isaac_sim/kit/python/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/noah/IsaacLab/_isaac_sim/kit/python/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/noah/.vscode/extensions/ms-python.debugpy-2025.0.1-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher/../../debugpy/__main__.py", line 71, in <module>
    cli.main()
  File "/home/noah/.vscode/extensions/ms-python.debugpy-2025.0.1-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher/../../debugpy/../debugpy/server/cli.py", line 501, in main
    run()
  File "/home/noah/.vscode/extensions/ms-python.debugpy-2025.0.1-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher/../../debugpy/../debugpy/server/cli.py", line 351, in run_file
    runpy.run_path(target, run_name="__main__")
  File "/home/noah/.vscode/extensions/ms-python.debugpy-2025.0.1-linux-x64/bundled/libs/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 310, in run_path
    return _run_module_code(code, init_globals, run_name, pkg_name=pkg_name, script_name=fname)
  File "/home/noah/.vscode/extensions/ms-python.debugpy-2025.0.1-linux-x64/bundled/libs/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 127, in _run_module_code
    _run_code(code, mod_globals, init_globals, mod_name, mod_spec, pkg_name, script_name)
  File "/home/noah/.vscode/extensions/ms-python.debugpy-2025.0.1-linux-x64/bundled/libs/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 118, in _run_code
    exec(code, run_globals)
  File "/home/noah/IsaacLab/scripts/reinforcement_learning/rsl_rl/train.py", line 13, in <module>
    from isaaclab.app import AppLauncher
  File "/home/noah/IsaacLab/source/isaaclab/isaaclab/__init__.py", line 9, in <module>
    import toml
ModuleNotFoundError: No module named 'toml'
```

### System Info

Describe the characteristic of your environment:

- Commit: f1a4975eb
- Isaac Sim Version: 4.5.0-rc.36+release.19112.f59b3005.gl
- OS: Ubuntu 20.04
- GPU: NVIDIA RTX A4500
- CUDA: 12.3
- GPU Driver: 545.23.08
- I have four GPU's

### Additional context

### Checklist

- [ ] I have checked that there is no similar issue in the repo - there are similar but all were closed with out an applicable solution
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] Criteria 1: am able to run and debug in vscode
- [ ] Criteria 2: the scripts generate the correct launch.json file

RandomOakForest, 2025-03-03:
Thank you for posting this. How did you install Isaac Lab and Isaac Sim? The VSCode setup is only available for binary installed Isaac Sim currently.

NNoah18, 2025-03-03:
I installed Isaac sim via binaries since that's what was recommended for Ubuntu 20.04. I started by following this page(https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/binaries_installation.html) but then started following the Isaac Sim installation page they linked(https://docs.isaacsim.omniverse.nvidia.com/latest/installation/install_workstation.html) before returning back to the first page, so the Isaac sim directory is not in the downloads folder, its in ~/isaacsim. I downloaded the Isaac sim and ran the following commands:
```
mkdir ~/isaacsim
cd ~/Downloads
unzip "isaac-sim-standalone@4.5.0-rc.36+release.19112.f59b3005.gl.linux-x86_64.release.zip" -d ~/isaacsim
cd ~/isaacsim
./post_install.sh
./isaac-sim.selector.sh
```
After running the scripts to verify the installation of Isaac sim, I preceded to install Isaac Lab as per the instructions on the page:
```
git clone git@github.com:isaac-sim/IsaacLab.git
# enter the cloned repository
cd IsaacLab
# create a symbolic link
ln -s ${HOME}/isaacsim _isaac_sim
```
I then skipped ```sudo apt install cmake build-essential``` because I verified that my system already had it installed, and would have had to ask someone else for sudo access.

and then I ran
```
./isaaclab.sh --install
```

then I ran the scripts to verify the installation of Isaac Lab - except the robot I first train was Isaac-Velocity-Flat-Spot-v0, and I was able to train it. I did also follow the following instructions:

"Open the IsaacLab directory on Visual Studio Code IDE

Run VSCode [Tasks](https://code.visualstudio.com/docs/editor/tasks), by pressing Ctrl+Shift+P, selecting Tasks: Run Task and running the setup_python_env in the drop down menu."

I also installed the "Isaac Sim VS Code Edition" VS Code extension.

closed 2025-03-05 by Mayankm96

NNoah18, 2025-03-06:
I manually edited the launch.json file to be correct, and from what I could tell, changing the launch.template.json as done in [#2021](https://github.com/isaac-sim/IsaacLab/pull/2021) only changes the "program" lines of the launch.json file to be what I manually did, and therefore I will still get the error I included in my original post.  So can issue be reopened since it is not solved and shouldn't be closed as complete?

EricPedley, 2025-07-30:
I am getting the `No module named 'toml'` error when I try running the debugger as well on Isaac Sim 5.0.0, installed from source. I think the PR that closed this issue only addresses the first point of this issue about the paths being wrong.

EricPedley, 2025-07-31:
I fixed it myself by adding the line `"python": "${workspaceFolder}/_isaac_sim/python.sh"` to my launch.json "Python: Current File" configuration. This is mentioned in this other issue here: https://github.com/isaac-sim/IsaacLab/issues/2238

PR #2021 (2025-03-05): Fixes path in the `launch.json`
# Description

Fix paths in `launch.json` to be adjusted to the new repo structure.

Fixes #1976

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with `./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
- [ ] I have added tests that prove my fix is effective or that my feature works
- [ ] I have updated the changelog and the corresponding version in the extension's `config/extension [... cut]
commit b8b79cfe (2025-03-05): Fixes path in the `launch.json` for RL training (#2021)

# Description

Fix paths in `launch.json` to be adjusted to the new repo structure.

Fixes #1976

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
- [ ] I have added tests that prove my fix is effective or that my
feature works
- [ ] I have updated the changelog and the corresponding version in the
exte [... cut]