isaac-sim/IsaacLab #1985: [Question] Can not open it correctly

gxy-TJU, 2025-03-02:
### Question
I am a beginner to use this.
And when I install it follow the document, there is no error. But when I run it with example
 `python scripts/tutorials/00_sim/create_empty.py`
some error happens, 

[image]

and there will be a window like this, and when I click wait, it will keep quiet, and I can not click any button.

[image]

My computer is a laptop, RTX 3070, 32G RAM, ubuntu 20.04.
Why this happen, and what can I do to fix this problem?
PS: I can run `./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Ant-v0`normal

[image]
Thanks for any response!

RandomOakForest, 2025-03-03:
Thank you for posting this. What happens when you run it using the following?
```
./isaaclab.sh -p scripts/tutorials/00_sim/create_empty.py
```
If you use Python directly or from an IDE, you need to be sure your Python environment has been properly set up. Be sure you can [verify the installation as explained here](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/binaries_installation.html).

gxy-TJU, 2025-03-06:
> Thank you for posting this. What happens when you run it using the following?
>
> ```
> ./isaaclab.sh -p scripts/tutorials/00_sim/create_empty.py
> ```
>
> If you use Python directly or from an IDE, you need [...]

No, I just run the command in the installation, and when I run `./isaaclab.sh -p scripts/tutorials/00_sim/create_empty.py`, it seems be normal.

gxy-TJU, 2025-03-06:
> > Thank you for posting this. What happens when you run it using the following?
> > ```
> > ./isaaclab.sh -p scripts/tutorials/00_sim/create_empty.py
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > If you use Python directly or from an IDE, you need to be sure your Python environment has been properly set up. Be sure you can [verify the installation as explained here](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/binaries_installation.html).
> 
>Thank you for your response, I just run the command in the installation, and when I run `./isaaclab.sh -p scripts/tutorials/00_sim/create_empty.py`,there still some errors, 

[image]

I do not know what this mean. But the gui seems be normal, I can add some entity like franka robot in it.

RandomOakForest, 2025-04-03:
Thanks for following up. This may point to missing libraries in the installation process. A newer issue #2110 is being used to track this problem. Will close this question for now. Thanks.

closed 2025-04-03 by RandomOakForest