isaac-sim/IsaacLab #1993: [Bug Report] RuntimeError: Sensor at path 'xxx' could not find any bodies with contact reporter API.

KevinGeng3, 2025-03-03:
### Describe the bug

When I use DirecteEnv for RL training, I add a cube to the scene:
```
    object_cfg = RigidObjectCfg(
        prim_path="/World/envs/env_.*/object",
        spawn=sim_utils.UsdFileCfg(
            activate_contact_sensors=True,
            usd_path="xxx/cube.usd",
            rigid_props=sim_utils.RigidBodyPropertiesCfg(
                rigid_body_enabled=True,
                kinematic_enabled=False,
                disable_gravity=False,
            ),
            scale=(1.0, 5.0, 2.0),
            mass_props=sim_utils.MassPropertiesCfg(mass=100.0),
        ),
        init_state=RigidObjectCfg.InitialStateCfg(pos=(0.25, 0.0, 0.78))
    )
```
And I add contact sensor on this cube:
```
object_sensor_cfg = ContactSensorCfg(
        prim_path="/World/envs/env_.*/object/.*",
        update_period=0.0,
        history_length=5,
        debug_vis=False,
    )
```
However, when I run the code, an error occurred:
```
Traceback (most recent call last):
  File "xxx/IsaacLab/source/isaaclab/isaaclab/sensors/sensor_base.py", line 66, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._initialize_callback(event),
  File "xxx/IsaacLab/source/isaaclab/isaaclab/sensors/sensor_base.py", line 273, in _initialize_callback
    self._initialize_impl()
  File "xxx/IsaacLab/source/isaaclab/isaaclab/sensors/contact_sensor/contact_sensor.py", line 260, in _initialize_impl
    raise RuntimeError(
RuntimeError: Sensor at path '/World/envs/env_.*/object/.*' could not find any bodies with contact reporter API.
HINT: Make sure to enable 'activate_contact_sensors' in the corresponding asset spawn configuration.
```
But I already enabled 'activate_contact_sensors'. Any help is appreciated!

### System Info

Describe the characteristic of your environment:

- Isaac Sim Version: 4.5
- Isaac Lab Version: 2.0
- OS: Ubuntu 22.04
- GPU: RTX3090
- CUDA: 12.4
- GPU Driver: 550.120

RandomOakForest, 2025-03-03:
Thanks for posting this. This is probably related to the `prim_path` regex you are using. [Following this example to add the sensor to your robot may help](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/04_sensors/add_sensors_on_robot.html#the-code). Let us know.

Mayankm96, 2025-03-05:
Does your USD file have colliders on it?

closed 2025-04-07 by RandomOakForest

hesic73, 2025-05-18:
I encountered the same problem when I tried to remove the hands from a Unitree G1 minimal USD file.

Jushan-Randy-Chen, 2025-06-13:
> Does your USD file have colliders on it?

I'm also having the same issue! I have set `activate_collision_sensor=True`  in `sim_utils.UsdFileCfg()`,but I'm still getting the "could not find any bodies with contact reporter API" error. Does anyone know how to fix this? thanks