isaac-sim/IsaacLab #1995: [Bug Report] [omni.physx.tensors.plugin] GPU contact filter for collider '/World/ground' is not supported

bikcrum, 2025-03-03:
### Describe the bug

I have defined my contact forces as follows:-

```
    contact_forces = ContactSensorCfg(
        prim_path="{ENV_REGEX_NS}/Robot/.*",
        filter_prim_paths_expr=["/World/ground"],
        history_length=3,
        track_air_time=True,
    )
```

I have verified that the prim path is correct

[image]

However, I get the following warning in the console, and the contact forces are not being registered.

```
 [Warning] [omni.physx.tensors.plugin] GPU contact filter for collider '/World/ground' is not supported
```

IsaacSim: 4.5.0
IsaacLab: 2.0.0

RandomOakForest, 2025-03-08:
Thank you for posting this. The team will review.

kellyguo11, 2025-03-09:
This is currently a limitation in the physics engine, where contacts on static colliders that are not rigid bodies are not yet supported. A possible workaround for now is to make the ground plane a rigid body by adding a RigidBodyAPI to the prim and enabling the kinematic attribute on it. Feel free to also check out https://isaac-sim.github.io/IsaacLab/main/source/how-to/make_fixed_prim.html for details on static colliders and kinematic bodies.

bikcrum, 2025-03-09:
> This is currently a limitation in the physics engine, where contacts on static colliders that are not rigid bodies are not yet supported. A possible workaround for now is to make the ground plane a [...]

I have a follow-up question. Why is this only a limitation when `filter_prim_paths_expr` is set? The collision with the ground works just fine when I don't use a filter.

kellyguo11, 2025-03-10:
The limitation is in the APIs in retrieving contacts on static colliders, thus the filtering logic also does not currently support them. However, the physics simulation should have no issues in solving for contacts on static colliders.

closed 2025-04-07 by RandomOakForest

bikcrum, 2025-05-13:
> hi [@bikcrum](https://github.com/bikcrum), how you make ground plan rigid body? can you provide code snippet.

@harshp1509 

I used a `RigidObjectCfg` inside the `SceneCfg` to define a flat ground plane. As far as I know, `TerrainImporterCfg` doesn’t support a `rigid_props` field, so I couldn’t use it directly for this purpose.

Below is a code snippet that worked for me. One thing to note: The size of the rigid body is large enough to fully cover the area beneath the assets otherwise, they may fall through due to a lack of collision support.

```
@configclass
class MySceneCfg(InteractiveSceneCfg):
  ...
  terrain = RigidObjectCfg(
      prim_path="/World/ground",
      spawn=sim_utils.CuboidCfg(
          size=(100.0, 100.0, 0.0001),
          physics_material=sim_utils.RigidBodyMaterialCfg(
              friction_combine_mode="multiply",
              restitution_combine_mode="multiply",
              static_friction=1.0,
              dynamic_friction=1.0,
          ),
          rigid_props=sim_utils.RigidBodyPropertiesCfg(kinematic_enabled=True),
          collision_props=sim_utils.CollisionPropertiesCfg(),
          visual_material=sim_utils.MdlFileCfg(
              mdl_path=(
                  f"{ISAACLAB_NUCLEUS_DIR}/Materials/TilesMarbleSpiderWhiteBrickBondHoned/"
                  "TilesMarbleSpiderWhiteBrickBondHoned.mdl"
              ),
              project_uvw=True,
              texture_scale=(0.25, 0.25),
          ),
      ),
      init_state=RigidObjectCfg.InitialStateCfg(pos=(0.0, 0.0, 0.0), rot=(1.0, 0.0, 0.0, 0.0)),
  )
  ...
```

harshp1509, 2025-05-14:
when i am trying to above way (mean i am getting prim name "/world/ground") than i am getting some cuda error but when i am using prim name "{ENV_REGEX_NS}/ground}" than it worked. but it is adding in every env. but i want to add only one time means universal.

```
import math
import torch
import os
from isaaclab.envs import ManagerBasedRLEnvCfg
from isaaclab.envs import ManagerBasedRLEnv
from isaaclab.managers import EventTermCfg as EventTerm
from isaaclab.managers import ObservationGroupCfg as ObsGroup
from isaaclab.managers import ObservationTermCfg as ObsTerm
from isaaclab.managers import RewardTermCfg as RewTerm
from isaaclab.managers import SceneEntityCfg
from isaaclab.managers import TerminationTermCfg as DoneTerm
from isaaclab.utils import configclass
from isaaclab.utils.assets import ISAAC_NUCLEUS_DIR, ISAACLAB_NUCLEUS_DIR
import isaaclab.sim as sim_utils

import isaaclab_tasks.manager_based.navigation.mdp as mdp
from isaaclab.assets import ArticulationCfg, AssetBaseCfg, DeformableObjectCfg, RigidObjectCfg
from isaaclab.actuators import ImplicitActuatorCfg
from isaaclab.sim.spawners.from_files.from_files_cfg import GroundPlaneCfg, UsdFileCfg
from isaaclab.scene import InteractiveSceneCfg
from isaaclab.sensors import RayCasterCfg, patterns
from isaaclab.sensors import ContactSensorCfg
from isaaclab.utils.noise import AdditiveUniformNoiseCfg as Unoise
from isaaclab.terrains import TerrainImporterCfg
from isaaclab.terrains.config.rough import ROUGH_TERRAINS_CFG 
from isaaclab.managers import CurriculumTermCfg as CurrTerm

JETBOT_CFG = ArticulationCfg(
    spawn=sim_utils.UsdFileCfg(
        usd_path="/media/airangers/AiRangersData2/Harsh/jetbot.usd",
        activate_contact_sensors=True,
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            rigid_body_enabled=True,
            max_linear_velocity=1000.0,
            max_angular_velocity=1000.0,
            max_depenetration_velocity=100.0,
            enable_gyroscopic_forces=True,
        ),
        articulation_props=sim_utils.ArticulationRootPropertiesCfg(
            enabled_self_collisions=False, 
            solver_position_iteration_count=4, 
            solver_velocity_iteration_count=0,
            sleep_threshold=0.005,
            stabilization_threshold=0.001,
        ),
    ),
    init_state=ArticulationCfg.InitialStateCfg(
        joint_pos={"right_wheel_joint": 0.0, "left_wheel_joint": 0.0},
    ),
    actuators={
        "wheel_motors": ImplicitActuatorCfg(
            joint_names_expr=["right_wheel_joint", "left_wheel_joint"],
            effort_limit=40000.0,
            velocity_limit=100.0,
            stiffness=0.0,
            damping=100000.0,
        ),
    },
)

@configclass
class JetSceneCfg(InteractiveSceneCfg):
    """Configuration for a Jet collision avoidance scene using LiDAR."""

    robot: ArticulationCfg = JETBOT_CFG.replace(prim_path="{ENV_REGEX_NS}/Robot")
    # Attach LiDAR to the jet

[... 196 lines cut ...]
    terminations: TerminationsCfg = TerminationsCfg()
    # curriculum: CurriculumCfg = CurriculumCfg()

    def __post_init__(self):
        """Post initialization."""

        self.decimation = 2
        self.sim.dt = 0.005  # simulation timestep -> 200 Hz physics
        self.episode_length_s = 5.0
        self.sim.render_interval = self.decimation
        self.sim.physx.gpu_max_rigid_patch_count = 10 * 2**15

        if self.scene.height_scanner is not None:
            self.scene.height_scanner.update_period = self.decimation * self.sim.dt

        if self.scene.contact_forces is not None:
            self.scene.contact_forces.update_period = self.sim.dt

        # if getattr(self.curriculum, "terrain_levels", None) is not None:
        #     if self.scene.terrain.terrain_generator is not None:
        #         self.scene.terrain.terrain_generator.curriculum = True
        # else:
        #     if self.scene.terrain.terrain_generator is not None:
        #         self.scene.terrain.terrain_generator.curriculum = False
```

bikcrum, 2025-05-18:
> "{ENV_REGEX_NS}/ground}" than it worked

It will be helpful if you can show the error logs.

harshp1509, 2025-05-19:
i am getting invalida mesh prim error when i am giving prim path as "/World/ground"

```
Traceback (most recent call last):
  File "/media/airangers/AiRangersData2/Harsh/IsaacLab/source/isaaclab/isaaclab/sensors/sensor_base.py", line 66, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._initialize_callback(event),
  File "/media/airangers/AiRangersData2/Harsh/IsaacLab/source/isaaclab/isaaclab/sensors/sensor_base.py", line 273, in _initialize_callback
    self._initialize_impl()
  File "/media/airangers/AiRangersData2/Harsh/IsaacLab/source/isaaclab/isaaclab/sensors/ray_caster/ray_caster.py", line 147, in _initialize_impl
    self._initialize_warp_meshes()
  File "/media/airangers/AiRangersData2/Harsh/IsaacLab/source/isaaclab/isaaclab/sensors/ray_caster/ray_caster.py", line 173, in _initialize_warp_meshes
    raise RuntimeError(f"Invalid mesh prim path: {mesh_prim_path}")
RuntimeError: Invalid mesh prim path: /World/ground
```

bikcrum, 2025-05-19:
@harshp1509 I see that you are using rigid body for terrain and naming it prim path to `"/World/defaultGroundPlane`. Does it throw any issues?

bikcrum, 2025-05-19:
@harshp1509 Instead of using `RigidObjectCfg` for terrain, you can try using `TerrainImporterCfg` with `MeshPlaneTerrainCfg`. See if you still get that error.

Try following:-
```
FLAT_TERRAINS_CFG = terrain_gen.TerrainGeneratorCfg(
    size=(8.0, 8.0),
    border_width=20.0,
    num_rows=9,
    num_cols=21,
    horizontal_scale=0.1,
    vertical_scale=0.005,
    slope_threshold=0.75,
    difficulty_range=(0.0, 1.0),
    use_cache=False,
    sub_terrains={
        "flat": terrain_gen.MeshPlaneTerrainCfg(proportion=1.0),
    },
)

@configclass
class MySceneCfg(InteractiveSceneCfg):
    """Configuration for the terrain scene with a legged robot."""

    # ground terrain
    terrain = TerrainImporterCfg(
        prim_path="/World/ground",
        terrain_type="generator",
        terrain_generator=FLAT_TERRAINS_CFG,
        max_init_terrain_level=0,
        collision_group=-1,
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="multiply",
            restitution_combine_mode="multiply",
            static_friction=1.0,
            dynamic_friction=1.0,
        ),
        visual_material=sim_utils.MdlFileCfg(
            mdl_path=f"{ISAACLAB_NUCLEUS_DIR}/Materials/TilesMarbleSpiderWhiteBrickBondHoned/TilesMarbleSpiderWhiteBrickBondHoned.mdl",
            project_uvw=True,
            texture_scale=(0.25, 0.25),
        ),
        debug_vis=False,
    )
    ...
```

harshp1509, 2025-05-19:
@bikcrum, now that error is not coming but in contact sensor "filter_prim_paths_expr" is not working mean it is not excluding the prim ("/world/ground"). whenever jetbot load it got terminate because it is also consider ground as contact.

bikcrum, 2025-05-20:
> [@bikcrum](https://github.com/bikcrum), now that error is not coming but in contact sensor "filter_prim_paths_expr" is not working mean it is not excluding the prim ("/world/ground"). whenever jetbot load it got terminate because it is also consider ground as contact.

But isn't that what you want? When you set `filter_prim_paths_expr`, you are registering a sensor between your body and that prim. If you don't want that, you should remove it from the filter and only put those prim where you need a collision check. 

Furthermore, if it is not the sensor alone that is causing the termination. You must have a termination function that terminates due to the collision. You can simply do a check there and terminate when your required condition is met.