isaac-sim/IsaacLab #2001: [Question] Arrived at the location but did not collect data.

happy233666, 2025-03-04:
Question 
When I used `./isaaclab.sh -p scripts/tools/record_demos.py` in the environment `Isaac-Lift-Cube-Franka-IK-Rel-v0` to prepare to collect data for imitation learning, the cube reached the specified position but did not proceed to the next round.
In the environment `Isaac-Stack-Cube-Franka-IK-Rel-v0`, it works normally.
Please tell me if I have any parameters to set or if the command is incorrect
[

[image]

](url)

happy233666, 2025-03-04:
I found out that there are no conditions for success, why

[image]

happy233666, 2025-03-04:
I found a workaround, just add “success” to the file below

[image]

nvcyc, 2025-03-04:
Yes, you are correct that a termination term named "success" must be defined as you suggested for the recorder to determine whether the task in the environment has been completed successfully or not for exporting the collected data.

closed 2025-03-04 by happy233666