isaac-sim/IsaacLab #2003: [Bug Report] motion_vectors / optical flow not aligned with other modalities (RGB, etc.)

lhy0807, 2025-03-04:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

I find the rendered motion_vectors (or optical flow) is not aligned with other modalities (such as RGB, depth, segmentation, etc.)

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

```
rgb = self.env.scene.sensors['tiled_camera'].data.output['rgb']
depth = self.env.scene.sensors['tiled_camera'].data.output['distance_to_camera']
semantic = self.env.scene.sensors['tiled_camera'].data.output['semantic_segmentation']
optical_flow = self.env.scene.sensors['tiled_camera'].data.output['motion_vectors']

# visualize rgb, depth, semantic and optical flow
fig, axs = plt.subplots(2, 2)
axs[0, 0].imshow(rgb[0].cpu().numpy())
axs[0, 0].set_title('rgb')
axs[0, 1].imshow(depth[0].cpu().numpy())
axs[0, 1].set_title('depth')
axs[1, 0].imshow(semantic[0].cpu().numpy())
axs[1, 0].set_title('semantic')
axs[1, 1].imshow(flow_vis.flow_to_color(optical_flow[0].cpu().numpy()))
axs[1, 1].set_title('optical flow')
plt.show()
```

[image]

### System Info

Describe the characteristic of your environment:

- Commit: 6a415df2
- Isaac Sim Version: 4.5.0-rc.36+release.19112.f59b3005.gl
- OS: Ubuntu 20.04
- GPU: RTX 4070 
- CUDA: 12.3
- GPU Driver: 545.23.06

### Additional context

Add any other context about the problem here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Find a way to align motion_vectors to other modalities

lhy0807, 2025-03-04:
[image]

Visualizing it in Isaac Sim seems fine

garylvov, 2025-03-04:
@kellyguo11

lhy0807, 2025-03-04:
From my preliminary testing, I think the problem might be related to this line

https://github.com/isaac-sim/IsaacLab/blob/4868e19c1df21715d6f58c03578d2ea0f29c7561/source/isaaclab/isaaclab/sensors/camera/tiled_camera.py#L262-L272

lhy0807, 2025-03-04:
I can confirm it's the `reshape_tiled_image` issue and find a quick fix to it.

After applying this, it works normally
```
# HACK: fix motion_vectors reshape issue by only feeding the first 2 channels
if data_type == "motion_vectors":
    tiled_data_buffer = tiled_data_buffer[:, :, :2].contiguous()

wp.launch(
    kernel=reshape_tiled_image,
    dim=(self._view.count, self.cfg.height, self.cfg.width),
    inputs=[
        tiled_data_buffer.flatten(),
        wp.from_torch(self._data.output[data_type]),  # zero-copy alias
        *list(self._data.output[data_type].shape[1:]),  # height, width, num_channels
        self._tiling_grid_shape()[0],  # num_tiles_x
    ],
    device=self.device,
)
```

[image]

PR #2012 (2025-03-05): Fixes misalignment in motion vectors from TiledCamera
# Description

I find the rendered motion_vectors (or optical flow) are not aligned with other modalities (such as RGB, depth, segmentation, etc.).
This commit is a workaround to fix this misalignment issue.

Fixes #2003

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Screenshots

Before:
[image]

After:
[image]

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with `./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documentation
- [x] My changes generate no  [... cut]
commit efa4e28b (2025-03-05): Fixes misalignment in motion vectors from TiledCamera (#2012)

# Description

I find the rendered motion_vectors (or optical flow) are not aligned
with other modalities (such as RGB, depth, segmentation, etc.).
This commit is a workaround to fix this misalignment issue.

Fixes #2003

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Screenshots

Before:

[image]

After:

[image]

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documentation
- [x] My changes g [... cut]
closed 2025-03-05 by Mayankm96