isaac-sim/IsaacLab #2004: [Proposal] Cloth and Fluid Simulation API

j3soon, 2025-03-04:
### Proposal

The current [soft body simulation](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/01_assets/run_deformable_object.html) works well. However, it appears that Cloth and Fluid simulations have not been introduced in Isaac Lab yet. I'm creating this issue to track the progress of these features and consolidate related discussions to prevent future confusion.

### Motivation

As mentioned in [the deformable object tutorial](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/01_assets/run_deformable_object.html), deformable objects include cloths, fluids, and soft bodies. Although soft bodies are supported in the current release (Isaac Lab v2.0.1), cloth and fluid simulation isn't officially supported through the Isaac Lab API yet.

Since [cloth simulation](https://youtu.be/wtpPxLmiykk) and [fluid simulation](https://youtu.be/eMyroevX1nA) is supported in Isaac Sim (i.e., `Particle Cloth` and `Particle Sampler` in `Window > Examples > Physics Examples` in Isaac Sim 4.5.0). There seems to be many confusion around whether Isaac Lab supports these features. See the issues/discussions related to this topic below:

- https://github.com/isaac-sim/IsaacLab/issues/748
- https://github.com/isaac-sim/IsaacLab/discussions/509
- https://github.com/isaac-sim/IsaacLab/discussions/1120
- https://github.com/isaac-sim/IsaacLab/discussions/1105
- https://github.com/isaac-sim/IsaacLab/discussions/587

After the introduction of the Cloth/Fluid API, I can help test them and improve the documentation and examples. Thank you!

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Cloth Simulation API and Examples
- [ ] Fluid Simulation API and Examples

Brayden-Zhang, 2025-03-05:
We are working on getting this integrated with @animesh-garg

AnnabellaMacaluso, 2025-05-15:
Hi are there any updates on the status of this integration? Thanks!

Brayden-Zhang, 2025-11-06:
Hi, I'm no longer working on this project anymore, so not sure what the status of it is.

Holger-AI, 2026-03-23:
Hi are there any updates on the status of this integration? Thanks! Holger