isaac-sim/IsaacLab #2015: [Question] Question title

nkbs33, 2025-03-05:
### Question

I tried AMP and found strange problems. When I finished training, and loaded the latest checkpoint, the robot immediately disappeared after the scene loading. But if I load the first saved checkpoint (agent_8000.pt), the scene works just fine. What could lead to this problem? Thank you.

I checked the output, there was no difference between these two cases, and no error was reported.

RandomOakForest, 2025-03-06:
Thank you for posting this. That sounds strange. I'll ask the team and get back to you. Is there any way you can share something for us to repro this? Thanks.

Toni-SM, 2025-03-06:
Hi @nkbs33 

A similar behavior was reported when the actions generated by the policy are unbounded (in that case, actions generated by random exploration in TD3 in Isaac Lab): https://github.com/Toni-SM/skrl/discussions/261#discussioncomment-11938793

Could you please provide more info about your evaluation, for example, the action values when applying it: https://github.com/isaac-sim/IsaacLab/blob/b5fa0eb031a2413c182eeb54fa3a9295e8fd867c/source/isaaclab_tasks/isaaclab_tasks/direct/humanoid_amp/humanoid_amp_env.py#L76

nkbs33, 2025-03-10:
I see. Indeed the problem should be numerical. I'll try to dump some value.

celestialdr4g0n, 2025-03-11:
You can try the following things:
- If you are working with manipulators, try clamp your actions in (-1,1)
- Lower your kl_threshold
You can spot if this happens or not by checking tensorboard, NaN reward is the indication.

nkbs33, 2025-03-11:
The loss goes to NaN as shown in tensorboard. I tried to limit the action space with gymnasium.spaces.Box,
but it didn't solve the problem. Is there another way to clamp the action? 

[image]

celestialdr4g0n, 2025-03-11:
I use torch.clamp to clamp my self.actions. I have never trail 4 legs robot before so I can not give you other suggestions.

nkbs33, 2025-03-11:
I tried clamping, but it does not elinimate this numerical instability. So sad.

Toni-SM, 2025-03-11:
Hi @nkbs33 

Could you please share agent config and TensorBoard image of "Policy" behavior (cut in your previous image)?

nkbs33, 2025-03-13:
Hi, I am running the humanoid_amp_run example without any change. The platform is win10.
[image]

qxjfly, 2025-03-18:
Same problem, when i run the `[Isaac-Humanoid-AMP-Run-Direct-v0]` example

qxjfly, 2025-03-19:
> Same problem, when i run the `[Isaac-Humanoid-AMP-Run-Direct-v0]` example
> 
> From the below video, the humanoid model in the environment has been disappear in 6500+ iterations, after that point, we play the 6500+.pt, the robot appeared once and immediately disappeared after the scene loading.
> 03181650.webm
> 03181709.webm

When I modified `HumanoidAmpEnvCfg`: `reset_strategy` from `random` to `random-start`, the problem disappeared.

Toni-SM, 2025-03-19:
Hi @nkbs33, @qxjfly 

Probably it is because, for some reason, when resetting the environment, the humanoid is in collision with the ground plane.
Currently the humanoid is a slightly elevated with respect to the ground, but this difference may not be sufficient:

https://github.com/isaac-sim/IsaacLab/blob/91f53e2fe5702eb2fd4bc4de98ab19d320f772c9/source/isaaclab_tasks/isaaclab_tasks/direct/humanoid_amp/humanoid_amp_env.py#L159

qxjfly, 2025-03-19:
> Hi [@nkbs33](https://github.com/nkbs33), [@qxjfly](https://github.com/qxjfly)
> 
> Probably it is because, for some reason, when resetting the environment using the `random` strategy, the _teleported_ humanoid is in collision with the ground plane. Currently the humanoid is a slightly elevated with respect to the ground, but this difference may not be sufficient:
> 
> [IsaacLab/source/isaaclab_tasks/isaaclab_tasks/direct/humanoid_amp/humanoid_amp_env.py](https://github.com/isaac-sim/IsaacLab/blob/91f53e2fe5702eb2fd4bc4de98ab19d320f772c9/source/isaaclab_tasks/isaaclab_tasks/direct/humanoid_amp/humanoid_amp_env.py#L159)
> 
> Line 159 in [91f53e2](/isaac-sim/IsaacLab/commit/91f53e2fe5702eb2fd4bc4de98ab19d320f772c9)
>  root_state[:, 2] += 0.15  # lift the humanoid slightly to avoid collisions with the ground

Hi @Toni-SM 
Thank you for your response！I retrained and played again after changing `root_state[:, 2] += 0.15` to `root_state[:, 2] += 0.25`, and the problem was solved.

nkbs33, 2025-03-20:
@Toni-SM @qxjfly Thank you for your help. I lifted root_state by 0.3 and the walking example runs correctly. But the running and dancing example still have this problem. Am I missing something?

celestialdr4g0n, 2025-03-20:
It looks like this is because invalid collision. You may want to observe the training process to make adjustments.

RandomOakForest, 2025-04-07:
Following up on this, it seems you got this working. Given the multiple issues that were covered here, it would be helpful to post a new question if you need further help in something more specific. I will close this issue for now. Thank you for your interest in Isaac Lab.

closed 2025-04-07 by RandomOakForest