isaac-sim/IsaacLab #2023: [Bug Report] RuntimeError: CUDA error: an illegal memory access was encountered

Leberkaesweckle, 2025-03-05:
### Describe the bug

I have already worked with Isaac-Sim 4.2 and IsaacLab. I have also installed Isaac-Sim 4.5 and IsaacLab 2.0 on a different PC, where they work fine. Now, I want to use them on this PC, but it's not working.

There is a direct crash at the launch of the first example:

``` 
./isaaclab.bat -p scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Ant-v0 --headless
```

results is: RuntimeError: CUDA error: an illegal memory access was encountered ...

I know that this is possibly not directly connected to IsaacLab, but maybe you have some ideas on how to solve it or have run into similar problems.
I have already tried it with different graphics drivers. I even tried it on Ubuntu 22.04, and this bug report is from a freshly reinstalled Windows 11. I am thankful for any help, I am a little bit desperate.
```
(isaaclab_4.5) PS C:\Users\ruedi\IsaacLab> ./isaaclab.bat -p scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Ant-v0 --headless
[INFO] Using python from: C:\Users\ruedi\miniconda3\envs\isaaclab_4.5\python.exe
[INFO][AppLauncher]: Loading experience file: C:\Users\ruedi\IsaacLab\apps\isaaclab.python.headless.kit
[Warning] [simulation_app.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [simulation_app.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: 'c:/users/ruedi/miniconda3/envs/isaaclab_4.5/lib/site-packages/omni/data/Kit/Isaac-Sim/4.5/user.config.json'
[Info] [carb] Logging to file: c:/users/ruedi/miniconda3/envs/isaaclab_4.5/lib/site-packages/omni/logs/Kit/Isaac-Sim/4.5/kit_20250305_122723.log
2025-03-05 11:27:23 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2025-03-05 11:27:24 [265ms] [Warning] [omni.usd_config.extension] Enable omni.materialx.libs extension to use MaterialX
2025-03-05 11:27:24 [723ms] [Warning] [omni.datastore] OmniHub is inaccessible
2025-03-05 11:27:24 [839ms] [Warning] [omni.isaac.dynamic_control] omni.isaac.dynamic_control is deprecated as of Isaac Sim 4.5. No action is needed from end-users.

|---------------------------------------------------------------------------------------------|
| Driver Version: 537.58        | Graphics API: D3D12
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090          | Yes: 0 |     | 24156   MB | 10de      | 8f090100.. |
|     |                                  |        |     |            | 2684      | 0          |
|     |                                  |        |     |            | 9         |            |
|=============================================================================================|
| OS: Windows 11 Pro, Version: 10.0 (24H2), Build: 26100, Kernel: 10.0.26100.2033
| Processor: AMD Ryzen 9 5900X 12-Core Processor
| Cores: 12 | Logical Cores: 24
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 65434 | Free Memory: 55487
| Total Page/Swap (MB): 74650 | Free Page/Swap: 60874
|---------------------------------------------------------------------------------------------|
[INFO]: Parsing configuration from: isaaclab_tasks.manager_based.classic.ant.ant_env_cfg:AntEnvCfg
[INFO]: Parsing configuration from: isaaclab_tasks.manager_based.classic.ant.agents.rsl_rl_ppo_cfg:AntPPORunnerCfg
[INFO] Logging experiment in directory: C:\Users\ruedi\IsaacLab\logs\rsl_rl\ant
Exact experiment name requested from command line: 2025-03-05_12-27-30
Setting seed: 42
[INFO]: Base environment:
        Environment device    : cuda:0
        Environment seed      : 42
        Physics step-size     : 0.008333333333333333
        Rendering step-size   : 0.016666666666666666
        Environment step-size : 0.016666666666666666
[INFO]: Time taken for scene creation : 2.607211 seconds
[INFO]: Scene manager:  <class InteractiveScene>
        Number of environments: 4096
        Environment spacing   : 5.0
        Source prim name      : /World/envs/env_0
        Global prim paths     : ['/World/ground']
        Replicate physics     : True
[INFO]: Starting the simulation. This may take a few seconds. Please wait...
[INFO]: Time taken for simulation start : 3.678103 seconds
[INFO] Command Manager:  <CommandManager> contains 0 active terms.
+------------------------+
|  Active Command Terms  |
+--------+-------+-------+
| Index  | Name  |  Type |
[... 261 lines cut ...]
- CUDA: 12.2
- GPU Driver: 537.58

[image]

### Additional context

IsaacSim is running perfect and the 
```
./isaaclab.bat -p scripts\tutorials\00_sim\create_empty.py
```
altough start without any other problems

Other ML frameworks also crash.
```
./isaaclab.bat -p scripts/reinforcement_learning/skrl/train.py --task=Isaac-Ant-v0 --headless  
```
altough crashes

### Checklist

- [ x ] I have checked that there is no similar issue in the repo (**required**)
- [ x ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Thank you very much for your help.

Leberkaesweckle, 2025-03-06:
So little startup on my Problem. 

I now installed it on a different PC: 

with a: Quadro RTX 4000 and Driver Version: 556.39

and on my Laptop

with a: RTX A3000 Laptop GPU and Driver Version: 553.24

both with Windows 11

and with bout it runs stable for 

So I cant even rep

If nobody else has some similar problem, I think this can be closed.

Leberkaesweckle, 2025-03-07:
Hey maybe someone has a idea how to solve, again working on the pc with the 4090 due the quadro is a little bit slow :D

RandomOakForest, 2025-04-18:
Thanks for following up. Please try again with the driver 552.86. If the issue persists, please file a new _Bug Report_. Thank you!

closed 2025-04-18 by RandomOakForest