isaac-sim/IsaacLab #2024: [Question] Speed-up render to real-time?

vassil-atn, 2025-03-05:
### Question
Is it possible to render the GUI at (near) real-time? In Isaac Gym this was possible, but with IsaacSim/Lab the rendering seems to be much slower than real-time even for a single agent on a 3060 - which makes performance evaluation difficult. Are there any settings I can play around with to get real-time inference?

hojae-io, 2025-03-05:
For a single agent, I usually run with `--device=cpu`. cpu has a faster rendering than GPU for single agent.

RandomOakForest, 2025-03-06:
Thanks for posting this. Have you tried the [suggested speed-ups in this doc](https://docs.isaacsim.omniverse.nvidia.com/latest/reference_material/sim_performance_optimization_handbook.html#isaac-sim-performance-optimization-handbook)?

tylerlum, 2025-04-02:
I can share some of my numbers for reference.

I am trying to run `play.py` to visualize my policy at real-time rates.

I added some prints to help understand where the bottleneck comes from:

```
        # time delay for real-time evaluation
        actual_dt = time.time() - start_time
        sleep_time = dt - actual_dt
        if args_cli.real_time:
            if sleep_time > 0:
                print(f"[INFO] Sleeping for {sleep_time*1000:.2f} ms, dt: {dt*1000:.2f} ms, actual dt: {actual_dt*1000:.2f} ms")
                time.sleep(sleep_time)
            else:
                print(f"[INFO] Real-time mode is not possible, dt: {dt*1000:.2f} ms, actual dt: {actual_dt*1000:.2f} ms")
```

## Headless

First, I compare cpu and gpu device when headless (no rendering, just physics and policy inference):

### GPU

```
--real-time --headless
```

~13-15ms for this alone
```
[INFO] Sleeping for 7.01 ms, dt: 20.00 ms, actual dt: 12.99 ms
timestep: 407
[INFO] Sleeping for 4.60 ms, dt: 20.00 ms, actual dt: 15.40 ms
timestep: 408
[INFO] Sleeping for 5.83 ms, dt: 20.00 ms, actual dt: 14.17 ms
timestep: 409
[INFO] Sleeping for 5.24 ms, dt: 20.00 ms, actual dt: 14.76 ms
timestep: 410
[INFO] Sleeping for 4.92 ms, dt: 20.00 ms, actual dt: 15.08 ms
timestep: 411
[INFO] Sleeping for 6.48 ms, dt: 20.00 ms, actual dt: 13.52 ms
timestep: 412
[INFO] Sleeping for 6.13 ms, dt: 20.00 ms, actual dt: 13.87 ms
timestep: 413
[INFO] Sleeping for 5.68 ms, dt: 20.00 ms, actual dt: 14.32 ms
timestep: 414
[INFO] Sleeping for 5.47 ms, dt: 20.00 ms, actual dt: 14.53 ms
```

### CPU

```
--real-time --device=cpu --headless
```

~4-13ms for this alone (CPU is clear winner)
```
[INFO] Sleeping for 7.51 ms, dt: 20.00 ms, actual dt: 12.49 ms
timestep: 99
[INFO] Sleeping for 15.06 ms, dt: 20.00 ms, actual dt: 4.94 ms
timestep: 100
[INFO] Sleeping for 7.09 ms, dt: 20.00 ms, actual dt: 12.91 ms
timestep: 101
[INFO] Sleeping for 9.83 ms, dt: 20.00 ms, actual dt: 10.17 ms
timestep: 102
[INFO] Sleeping for 12.88 ms, dt: 20.00 ms, actual dt: 7.12 ms
timestep: 103
[INFO] Sleeping for 15.24 ms, dt: 20.00 ms, actual dt: 4.76 ms
timestep: 104
[INFO] Sleeping for 14.77 ms, dt: 20.00 ms, actual dt: 5.23 ms
[... 101 lines cut ...]
timestep: 204
[INFO] Sleeping for 7.17 ms, dt: 20.00 ms, actual dt: 12.83 ms
timestep: 205
[INFO] Sleeping for 6.46 ms, dt: 20.00 ms, actual dt: 13.54 ms
timestep: 206
[INFO] Real-time mode is not possible, dt: 20.00 ms, actual dt: 23.01 ms
timestep: 207
[INFO] Sleeping for 6.55 ms, dt: 20.00 ms, actual dt: 13.45 ms
timestep: 208
[INFO] Sleeping for 7.54 ms, dt: 20.00 ms, actual dt: 12.46 ms
timestep: 209
[INFO] Sleeping for 6.41 ms, dt: 20.00 ms, actual dt: 13.59 ms
timestep: 210
[INFO] Sleeping for 5.01 ms, dt: 20.00 ms, actual dt: 14.99 ms
timestep: 211
[INFO] Sleeping for 3.14 ms, dt: 20.00 ms, actual dt: 16.86 ms
timestep: 212
[INFO] Sleeping for 3.02 ms, dt: 20.00 ms, actual dt: 16.98 ms
```

Didn't need to do any adjustment to rendering settings with these tests

Related issues:

* #2230: Related because initially, the `--real-time` flag was using `physics_dt` instead of `step_dt`, so it was saying the desired dt should be `5ms` instead of `20ms` to account for `decimation = 4`. This was way too hard to reach performance-wise

closed 2025-04-07 by RandomOakForest