isaac-sim/IsaacLab #2028: [Bug Report] Problem when installing isaaclab

nadia-97, 2025-03-06:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

I am not sure if it is a bug or not. but I am facing this error when installing isaaclab, I tried installing some frameworks separately and that works but this command from the steps to install is giving me an error when installing rl-framework(isaaclab.bat --install :: or "isaaclab.bat -i"): 

ERROR: Exception:
Traceback (most recent call last):
  File "C:\Users\xxx\Documents\IsaacLab\_isaac_sim\kit\python\lib\site-packages\pip\_vendor\packaging\requirements.py", line 102, in __init__
    req = REQUIREMENT.parseString(requirement_string)
  File "C:\Users\xxx\Documents\IsaacLab\_isaac_sim\kit\python\lib\site-packages\pip\_vendor\pyparsing\util.py", line 256, in _inner
    return fn(self, *args, **kwargs)
  File "C:\Users\xxx\Documents\IsaacLab\_isaac_sim\kit\python\lib\site-packages\pip\_vendor\pyparsing\core.py", line 1190, in parse_string
    raise exc.with_traceback(None)
pip._vendor.pyparsing.exceptions.ParseException: Expected string_end, found '['  (at char 11), (line:1, col:12)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\xxx\Documents\IsaacLab\_isaac_sim\kit\python\lib\site-packages\pip\_internal\cli\base_command.py", line 180, in exc_logging_wrapper
    status = run_func(*args)
  File "C:\Users\xxx\Documents\IsaacLab\_isaac_sim\kit\python\lib\site-packages\pip\_internal\cli\req_command.py", line 245, in wrapper
    return func(self, options, args)
  File "C:\Users\xxx\Documents\IsaacLab\_isaac_sim\kit\python\lib\site-packages\pip\_internal\commands\install.py", line 342, in run
    reqs = self.get_requirements(args, options, finder, session)
  File "C:\Users\xxx\Documents\IsaacLab\_isaac_sim\kit\python\lib\site-packages\pip\_internal\cli\req_command.py", line 422, in get_requirements
    req_to_add = install_req_from_editable(
  File "C:\Users\xxx\Documents\IsaacLab\_isaac_sim\kit\python\lib\site-packages\pip\_internal\req\constructors.py", line 240, in install_req_from_editable
    parts = parse_req_from_editable(editable_req)
  File "C:\Users\xxx\Documents\IsaacLab\_isaac_sim\kit\python\lib\site-packages\pip\_internal\req\constructors.py", line 209, in parse_req_from_editable
    name, url, extras_override = parse_editable(editable_req)
  File "C:\Users\xxx\Documents\IsaacLab\_isaac_sim\kit\python\lib\site-packages\pip\_internal\req\constructors.py", line 112, in parse_editable
    get_requirement("placeholder" + extras.lower()).extras,
  File "C:\Users\xxx\Documents\IsaacLab\_isaac_sim\kit\python\lib\site-packages\pip\_internal\utils\packaging.py", line 45, in get_requirement
    return Requirement(req_string)
  File "C:\Users\xxx\Documents\IsaacLab\_isaac_sim\kit\python\lib\site-packages\pip\_vendor\packaging\requirements.py", line 104, in __init__
    raise InvalidRequirement(
pip._vendor.packaging.requirements.InvalidRequirement: Parse error at "'[::]'": Expected string_end

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by `cat ${ISAACSIM_PATH}/VERSION`]
- OS: [e.g. Ubuntu 20.04]
- GPU: [e.g. RTX 2060 Super]
- CUDA: [e.g. 11.4]
- GPU Driver: [e.g. 470.82.01, this can be seen by using `nvidia-smi` command.]

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-03-06:
Thanks for posting this. Could you try the commit of the latest version of Isaac Lab? That would be [2.0.2](https://github.com/isaac-sim/IsaacLab/releases/tag/v2.0.2). This may have been fixed. Thanks.

nadia-97, 2025-03-06:
Hi! Thanks for your reply. I already tried that today but same problem

huhonglei, 2025-03-18:
hi, i met the same problem。

RandomOakForest, 2025-04-07:
Following up, we are still unable to reproduce this. Could you share the doc you are following to install this? Thanks.

Sramez, 2025-04-14:
Getting the same problem, this is the document I'm following this document:
https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html#installing-isaac-lab

ERROR: Exception:
Traceback (most recent call last):
  File "C:\Users\xxx\miniconda3\envs\env_isaaclab\lib\site-packages\pip\_vendor\packaging\requirements.py", line 36, in __init__
    parsed = _parse_requirement(requirement_string)
  File "C:\Users\xxx\miniconda3\envs\env_isaaclab\lib\site-packages\pip\_vendor\packaging\_parser.py", line 62, in parse_requirement
    return _parse_requirement(Tokenizer(source, rules=DEFAULT_RULES))
  File "C:\Users\xxx\miniconda3\envs\env_isaaclab\lib\site-packages\pip\_vendor\packaging\_parser.py", line 77, in _parse_requirement
    extras = _parse_extras(tokenizer)
  File "C:\Users\xxx\miniconda3\envs\env_isaaclab\lib\site-packages\pip\_vendor\packaging\_parser.py", line 164, in _parse_extras
    with tokenizer.enclosing_tokens(
  File "C:\Users\xxx\miniconda3\envs\env_isaaclab\lib\contextlib.py", line 142, in __exit__
    next(self.gen)
  File "C:\Users\xxx\miniconda3\envs\env_isaaclab\lib\site-packages\pip\_vendor\packaging\_tokenizer.py", line 189, in enclosing_tokens
    self.raise_syntax_error(
  File "C:\Users\xxx\miniconda3\envs\env_isaaclab\lib\site-packages\pip\_vendor\packaging\_tokenizer.py", line 167, in raise_syntax_error
    raise ParserSyntaxError(
pip._vendor.packaging._tokenizer.ParserSyntaxError: Expected matching RIGHT_BRACKET for LEFT_BRACKET, after extras
    placeholder[::]
               ~^

nadia-97, 2025-04-14:
> Following up, we are still unable to reproduce this. Could you share the doc you are following to install this? Thanks.

https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html#installing-isaac-lab 
I am using this documentation to install as well

Sramez, 2025-04-15:
It was a silly mistake! I was running the whole {isaaclab.bat --install :: or "isaaclab.bat -i" }
You only need to run isaaclab.bat --install 
@RandomOakForest I guess you can change the document that when you copy the command only copies isaaclab.bat --install :)

Mercy2Green, 2025-05-07:
> It was a silly mistake! I was running the whole {isaaclab.bat --install :: or "isaaclab.bat -i" } You only need to run isaaclab.bat --install [@RandomOakForest](https://github.com/RandomOakForest) I [...]

This is the correct answer for me!!!! Thanks a lot.

closed 2025-07-07 by kellyguo11