isaac-sim/IsaacLab #2029: [bug] Ray Caster Does Not Move with the Robot in Isaac Lab

MiraiMoe, 2025-03-06:
## **Describe the bug**  
I encountered an issue when using **Isaac Lab's Ray Caster**: it does not move along with the robot. I enabled `debug_vis` and observed that the **point cloud remains static**, even though my robot moves as expected.  

[image]

My robot is modified from the official **cf2x** (as shown in the screenshot). I can control it properly using API functions. However, while the **camera moves with the robot, the Ray Caster remains fixed**.  

[image]

## **Steps to reproduce**  
1. Extract and use the provided model: crazyflie_final_new.zip.
2. Replace the usd_path in my code with:
```python
usd_path=r"/home/username/crazyflie_final_new.usd"
```
3. Run the provided script and observe the issue.
4. Enable debug_vis and notice that the Ray Caster's point cloud does not move, while the robot moves correctly.
5. The camera follows the UAV correctly, but the Ray Caster remains static.

## **Sensor Configuration**  
Here is my sensor configuration:

```python
camera = CameraCfg(
    prim_path="{ENV_REGEX_NS}/Robot/crazyflie/body/front_cam",
    update_period=0.1,
    height=480,
    width=640,
    data_types=["rgb", "distance_to_image_plane"],
    spawn=sim_utils.PinholeCameraCfg(
        focal_length=24.0, focus_distance=400.0, horizontal_aperture=20.955, clipping_range=(0.1, 1.0e5)
    ),
    offset=CameraCfg.OffsetCfg(pos=(0.15, 0.0, 0.15), rot=(-0.707, 0.707, 0, 0), convention="world"),
)

height_scanner = RayCasterCfg(
    prim_path="{ENV_REGEX_NS}/Robot/crazyflie/body",
    update_period=1 / 60,
    offset=RayCasterCfg.OffsetCfg(pos=(0, 0, 0)),
    attach_yaw_only=True,
    pattern_cfg=patterns.LidarPatternCfg(
        channels=100, vertical_fov_range=[-90, 90], horizontal_fov_range=[-90, 90], horizontal_res=1.0
    ),
    debug_vis=True,
    mesh_prim_paths=["/World/ground"],
)

```

---

## **Expected Behavior**  
The Ray Caster should move along with the robot, similar to the camera.

---

## **System Info**  
- **Isaac Sim Version**: [4.5.0]  
- **OS**: [e.g. Ubuntu 22.04]  
- **GPU**: [e.g. RTX 4070Ti]  
- **CUDA**: [e.g. 12.4]  
- **GPU Driver**: [e.g. 550.120]  

---
## **Code to Reproduce**
Here is my full code that demonstrates the issue, I wrote this code based on the tutorial of Isaac Lab:
```python
# Copyright (c) 2022-2025, The Isaac Lab Project Developers.
# All rights reserved.
#
[... 202 lines cut ...]
    # run the main function
    main()
    # close sim app
    simulation_app.close()
```

[crazyflie_final_new.zip](https://github.com/user-attachments/files/19104810/crazyflie_final_new.zip)
---

## **Additional context**  
- The **Ray Caster is attached to** `"/World/envs/env_.*/Robot/crazyflie/body"` but does not inherit motion.  
- The **camera follows the UAV correctly**, but the **Ray Caster remains fixed**.  
- I suspect the issue might be due to **how the Ray Caster's transformation is updated**.  

---

## **Checklist**  
- [x] I have checked that there is no similar issue in the repo (**required**)  
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo  

---

## **Acceptance Criteria**  
- [ ] The Ray Caster follows the robot’s movement correctly  
- [ ] The issue is reproduced and resolved in an upcoming update

RandomOakForest, 2025-03-06:
Thank you for posting this. It may be something missing in the sensor/robot configuration. The team will review.

MiraiMoe, 2025-03-12:
A few days ago, I encountered an issue in Isaac Lab where the raycaster attached to my UAV did not move with the UAV body—the points remained static. At that time, I checked the perspective of the camera attached to the UAV in Isaac Sim, and it appeared to move correctly with the UAV. Based on this observation, I initially suspected that the problem was with the raycaster.

However, today I inspected the depth images obtained using the Isaac Lab API and visualized them as grayscale images. To my surprise, I found that the depth camera also did not move as expected. This suggests that the issue might not be limited to the raycaster but could be related to my UAV setup.

Could this be a problem with my robot configuration? Any insights would be greatly appreciated.

tccoin, 2025-12-06:
Hi I am having the same issue here but with isaac sim 5.0 and isaac lab 2.2.0. Can any one help me to check how to solve it? I have tried the solution @MiraiMoe provided above but it's not working in my case.

I notice that even though the robot moves, the coordinate of base doesn't move in the gui. Is it because the translate of xform is recorded in usd and root_state_w is only updated on gpu fabric? @RandomOakForest 

[image]

This can be reproduced by this code. I modified the original demo to move the robot with write_root_pose_to_sim():

```
# Copyright (c) 2022-2025, The Isaac Lab Project Developers (https://github.com/isaac-sim/IsaacLab/blob/main/CONTRIBUTORS.md).
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

import argparse

from isaaclab.app import AppLauncher

# add argparse arguments
parser = argparse.ArgumentParser(description="Example on using the raycaster sensor.")
parser.add_argument("--num_envs", type=int, default=1, help="Number of environments to spawn.")
# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()

# launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

"""Rest everything follows."""

import numpy as np
import torch

import isaaclab.sim as sim_utils
from isaaclab.assets import AssetBaseCfg
from isaaclab.scene import InteractiveScene, InteractiveSceneCfg
from isaaclab.sensors.ray_caster import RayCasterCfg, patterns
from isaaclab.utils import configclass
from isaaclab.utils.assets import ISAAC_NUCLEUS_DIR

##
# Pre-defined configs
##
from isaaclab_assets.robots.anymal import ANYMAL_C_CFG  # isort: skip

@configclass
class RaycasterSensorSceneCfg(InteractiveSceneCfg):
    """Design the scene with sensors on the robot."""

    # ground plane
    ground = AssetBaseCfg(
        prim_path="/World/Ground",
        spawn=sim_utils.UsdFileCfg(
            usd_path=f"{ISAAC_NUCLEUS_DIR}/Environments/Terrains/rough_plane.usd",
            scale=(1, 1, 1),
        ),
    )

    # lights
    dome_light = AssetBaseCfg(
        prim_path="/World/Light", spawn=sim_utils.DomeLightCfg(intensity=3000.0, color=(0.75, 0.75, 0.75))
    )

    # robot
    robot = ANYMAL_C_CFG.replace(prim_path="{ENV_REGEX_NS}/Robot")

    ray_caster = RayCasterCfg(
        prim_path="{ENV_REGEX_NS}/Robot/base/lidar_cage",
[... 83 lines cut ...]
    """Main function."""

    # Initialize the simulation context
    sim_cfg = sim_utils.SimulationCfg(dt=0.005, device=args_cli.device)
    sim = sim_utils.SimulationContext(sim_cfg)
    # Set main camera
    sim.set_camera_view(eye=[3.5, 3.5, 3.5], target=[0.0, 0.0, 0.0])
    # design scene
    scene_cfg = RaycasterSensorSceneCfg(num_envs=args_cli.num_envs, env_spacing=2.0)
    scene = InteractiveScene(scene_cfg)
    # Play the simulator
    sim.reset()
    # Now we are ready!
    print("[INFO]: Setup complete...")
    # Run the simulator
    run_simulator(sim, scene)

if __name__ == "__main__":
    # run the main function
    main()
    # close sim app
    simulation_app.close()
```

Thank you!!