isaac-sim/IsaacLab #2031: [Question] Correcting Camera Intrinsics for Accurate Sim2Real Alignment in Isaac Lab

Entongsu, 2025-03-06:
I am working on a Sim2Real transfer project and have the intrinsic and extrinsic parameters of my real-world camera. However, I am noticing a misalignment between the real and simulated images in Isaac Lab, which I suspect is due to incorrect intrinsic parameters.

I would like to know:

How can I accurately determine the focal length and focus distance values for the Isaac Lab camera?
I have noticed that slight variations in focal length can significantly impact the alignment, so any guidance on correctly setting these values would be greatly appreciated.

Additional Context:

I can access my real camera’s intrinsic matrix (K) and distortion coefficients.
The extrinsic transformation appears accurate, so the issue lies in the intrinsic parameters.

RandomOakForest, 2025-03-06:
Thanks for posting this. You'll have to work with [Replicator](https://docs.omniverse.nvidia.com/extensions/latest/ext_omnigraph/node-library/nodes/omni-replicator-core/cameraparams-1.html). The example [here is a good starting point](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/04_sensors/add_sensors_on_robot.html#camera-sensor).

Duuuc, 2025-03-06:
You can also create a PinholeCameraCfg for your Isaac Lab camera from an intrinsic matrix: https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.sim.spawners.html#isaaclab.sim.spawners.sensors.PinholeCameraCfg.from_intrinsic_matrix

edubya, 2025-03-22:
I've been setting intrinsics as in test/sensors/test_ray_caster_camera.py. 
```
        camera_cfg_zero = RayCasterCameraCfg(
            prim_path="/World/CameraZero",
            mesh_prim_paths=["/World/defaultGroundPlane"],
            offset=RayCasterCameraCfg.OffsetCfg(
                pos=(2.5, 2.5, 6.0), rot=(0.9914449, 0.0, 0.1305, 0.0), convention="world"
            ),
            pattern_cfg=patterns.PinholeCameraPatternCfg().from_intrinsic_matrix(
                focal_length=38.0,
                intrinsic_matrix=[380.08, 0.0, 467.79, 0.0, 380.08, 262.05, 0.0, 0.0, 1.0],
                height=540,
                width=960,
            ),
            max_distance=10.0,
            data_types=["distance_to_image_plane", "distance_to_camera"],
            depth_clipping_behavior="zero",
        )
```

closed 2025-04-07 by RandomOakForest