isaac-sim/IsaacLab #2032: [Question] Implementing Efficient Inertial Randomization: Episode Truncation and CPU-GPU Overhead Reduction

ParlitsisG, 2025-03-06:
### Question

I am training an RL agent and I want to incorporate mass, center of mass (COM), and inertia randomization, which must be performed on the CPU. My concern is the overhead of transferring tensors between the CPU and GPU every timestep when updating these parameters for different environments.

To address this, I am considering truncating all ongoing episodes and resetting all environments simultaneously, applying inertial randomization for all envs togehter instead of applying it for every env when it resets.

 However, Direct RL does not provide a built-in mechanism for truncating all episodes at once. How can I implement this functionality? Additionally, I want to verify whether this approach would indeed reduce overhead compared to transferring tensors to the CPU and back to the GPU every env at different timesteps.

Lastly, if I precompute the randomized mass, COM, and inertia values in a separate thread and store them in a CPU tensor, would using these precomputed values to directly set the mass COM and inertia values would further reduce overhead?

I appreciate any insights on the feasibility and implementation of this approach.

Thanks!

RandomOakForest, 2025-03-07:
Thank you for posting this. The team is discussing about this. We may have to move this post into a _discussion_.

RandomOakForest, 2025-03-14:
Following up, regarding truncating the episodes, you can truncate everything by just forcing the reset buffer to `True` for all environments `get_done()`s, and that will reset all environments together.

For the second part, caching all values used for reset on the CPU should help.

StrainFlow, 2025-03-14:
Just wanted to add that if you use pre-computed random values that would eliminate your bottleneck and let you terminate episodes asynchronously like normal.

closed 2025-04-07 by RandomOakForest

varadVaidya, 2025-05-27:
Hello! Sorry for commenting on a closed thread. @RandomOakForest can you share a sample code on what you mean by pre-computed random values. I am interested in changing the inertial parameters of a quadrotor at episode reset, and want to see how feasible this is. 
Minimal overhead is important to me, as i want to switch my current codebase from MuJoCo to Issac for faster simulation and training times. If the overhead is too costly in terms of wallclock time for RL training, it is better to stick with MuJoCo

StrainFlow, 2025-07-28:
What I meant by pre-computed random values is that you could compute a large collection of random values that you use instead of generating random numbers on the fly.