isaac-sim/IsaacLab #2035: [Question] if the .usd file property will be overwrite by the definition with DeformableObject/RigidObject

xinyu-66, 2025-03-07:
### Question

Hi, when I am loading the customized object into the isaac lab, we should define the RigidObject/Articulation/DeformableObject and with the hyparameters: `deformable_props`, `visual_material`, `physics_material`.

My question is that, I already have the .usd file with all these parameters settled in the isaacsim, when I loading them, if:
1. The definition in this IsaacLab API will overwrite the .usd file parameters.
2. I set the parameters to `None`, if they will set all the parameters to `None` or keep the configuration in the .usd file.

RandomOakForest, 2025-03-07:
Thank you for posting this. Could you share an example with code of how you _design the scene_? You can change parameters of your robot instances before you run the simulation, [for example](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/01_assets/run_articulation.html).

Mayankm96, 2025-03-07:
These parameters can be set only for cases where you'd like to modify those properties. Otherwise, if you keep them None, they'd stay at their USD defaults.

xinyu-66, 2025-03-10:
> Thank you for posting this. Could you share an example with code of how you _design the scene_? You can change parameters of your robot instances before you run the simulation, [for example](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/01_assets/run_articulation.html).

Thanks for your reply and Mayankm's answer solves my question.

xinyu-66, 2025-03-10:
> These parameters can be set only for cases where you'd like to modify those properties. Otherwise, if you keep them None, they'd stay at their USD defaults.

Thanks, I understand it right now. 

If we have the spatial tendon introduced in IsaacLab, I just see fixed tendon API in source code:
https://github.com/isaac-sim/IsaacLab/blob/main/source/isaaclab/isaaclab/sim/schemas/schemas_cfg.py

kellyguo11, 2025-03-10:
We are working on integrating spatial tendon APIs, please stay tuned!

xinyu-66, 2025-03-10:
> We are working on integrating spatial tendon APIs, please stay tuned!

Thanks! I will keep track.

closed 2025-04-07 by RandomOakForest