isaac-sim/IsaacLab #2036: [Question] GUI mode extremely slow or stuck with high cpu usage

changhoji, 2025-03-07:
### Question

After updating to IsaacLab 2.0, runing any environment in GUI mode (without `--headless`) causes extreme slowness or the application to become completely stuck. CPU usage reaches 100% while GPU remains low.

Even when running simple tutorial environments like `tutorials/00_sim/create_empty.py`, the issue persists. In `--headless` mode, it runs fine with GPU acceleration. This issue did not occur before the update.

Is my computer not powerful enough to run IsaacSim or IsaacLab properly? Could this issue can be caused by insufficient vram(8GB)? Are there any workarounds to prevent the simulation from getting stuck in GUI mode?

Any help or suggestions would be greatly appreciated. Thank you!

### System info
- Isaaclab Version: 2.0.1
- IsaacSim Version: 4.5.0
- CPU: Intel i7-9700k
- GPU: RTX 4060 Ti (8GB VRAM)
- OS: Windows 11
- RAM: 32GB
- CUDA Version: 12.8
- GPU Driver Version: 572.16

### Log

Below is the log output when running `./isaaclab.bat -p scripts/tutorials/00_sim/create_empty.py`

```./isaaclab.bat -p scripts/tutorials/00_sim/create_empty.py
[INFO] Using python from: C:\Users\onkkn\anaconda3\envs\env_isaaclab\python.exe
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: C:\Users\onkkn\dev\zolp\IsaacLab\apps\isaaclab.python.kit
[Warning] [simulation_app.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [simulation_app.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: 'c:/users/onkkn/anaconda3/envs/env_isaaclab/lib/site-packages/omni/data/Kit/Isaac-Sim/4.5/user.config.json'
[Info] [carb] Logging to file: c:/users/onkkn/anaconda3/envs/env_isaaclab/lib/site-packages/omni/logs/Kit/Isaac-Sim/4.5/kit_20250307_214633.log
2025-03-07 12:46:33 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.287s] [ext: omni.kit.async_engine-0.0.1] startup
[0.745s] [ext: omni.metrics.core-0.0.1] startup
[0.747s] [ext: omni.client.lib-1.0.0] startup
[0.800s] [ext: omni.blobkey-1.1.2] startup
[0.803s] [ext: omni.stats-1.0.1] startup
[0.809s] [ext: omni.datastore-0.0.0] startup
[0.815s] [ext: omni.client-1.2.2] startup
[0.830s] [ext: omni.ujitso.default-1.0.0] startup
[0.838s] [ext: omni.hsscclient-1.1.1] startup
2025-03-07 12:46:33 [736ms] [Warning] [omni.datastore] OmniHub is inaccessible
[0.853s] [ext: omni.gpu_foundation.shadercache.vulkan-1.0.0] startup
[0.867s] [ext: omni.assets.plugins-0.0.0] startup
[0.874s] [ext: omni.gpu_foundation-0.0.0] startup
[0.946s] [ext: carb.windowing.plugins-1.0.0] startup
[1.145s] [ext: omni.kit.renderer.init-0.0.0] startup
[1.148s] [ext: omni.kit.pipapi-0.0.0] startup
[1.164s] [ext: omni.kit.pip_archive-0.0.0] startup
[1.165s] [ext: omni.materialx.libs-1.0.6] startup
[1.212s] [ext: omni.kit.telemetry-0.5.1] startup
[1.240s] [ext: omni.usd.config-1.0.5] startup
[1.263s] [ext: omni.gpucompute.plugins-0.0.0] startup
[1.267s] [ext: omni.usd.libs-1.0.1] startup
[1.540s] [ext: omni.mdl-55.0.1] startup
[1.576s] [ext: omni.iray.libs-0.0.0] startup
[1.629s] [ext: omni.mdl.neuraylib-0.2.10] startup
[1.640s] [ext: omni.kit.usd.mdl-1.0.7] startup
[1.830s] [ext: omni.kit.loop-isaac-1.2.1] startup
[1.839s] [ext: omni.kit.test-1.1.2] startup
[2.063s] [ext: omni.appwindow-1.1.9] startup
[2.094s] [ext: omni.kit.renderer.core-1.0.2] startup
[2.110s] [ext: omni.kit.renderer.capture-0.0.0] startup
[2.123s] [ext: omni.kit.renderer.imgui-1.0.2] startup
[2.167s] [ext: omni.ui-2.26.5] startup
[2.197s] [ext: omni.kit.mainwindow-1.0.3] startup
[2.201s] [ext: carb.audio-0.1.0] startup
[2.203s] [ext: omni.uiaudio-1.0.0] startup
[2.214s] [ext: omni.kit.uiapp-0.0.0] startup
[... 245 lines cut ...]
[10.829s] [ext: omni.usd.metrics.assembler.ui-106.5.0] startup
[10.844s] [ext: isaacsim.asset.browser-1.3.4] startup
[11.387s] [ext: isaacsim.gui.menu-2.0.9] startup
2025-03-07 12:46:44 [11,297ms] [Warning] [omni.kit.menu.utils.app_menu] add_menu_items: menu [<MenuItemDescription name:'New'>, <MenuItemDescription name:'Open'>, <MenuItemDescription name:'Re-open with New Edit Layer'>, <MenuItemDescription name:'Save'>, <MenuItemDescription name:'Save With Options'>, <MenuItemDescription name:'Save As...'>, <MenuItemDescription name:'Save Flattened As...'>, <MenuItemDescription name:'Add Reference'>, <MenuItemDescription name:'Add Payload'>, <MenuItemDescription name:'Exit'>] cannot change delegate
[11.409s] [ext: omni.kit.ui.actions-1.0.2] startup
[11.419s] [ext: omni.kit.window.console-0.2.14] startup
[11.433s] [ext: isaacsim.robot_motion.motion_generation-8.0.6] startup
[11.454s] [ext: isaaclab-0.34.5] startup
[12.720s] [ext: isaaclab_assets-0.2.1] startup
[13.391s] [ext: isaaclab_tasks-0.10.24] startup
C:\Users\onkkn\anaconda3\envs\env_isaaclab\lib\site-packages\h5py\__init__.py:36: UserWarning: h5py is running against HDF5 1.14.4 when it was built against 1.14.6, this may cause problems
  _warn(("h5py is running against HDF5 {0} when it was built against {1}, "
[14.268s] [ext: omni.kit.menu.common-1.1.9] startup
[14.273s] [ext: isaaclab_rl-0.1.0] startup
[16.276s] [ext: isaaclab_mimic-1.0.2] startup
[16.278s] [ext: isaaclab.python-2.0.1] startup
[16.282s] Simulation App Starting
[16.728s] app ready
2025-03-07 12:46:49 [16,683ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit buffer size 7508933632
2025-03-07 12:46:49 [16,684ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : valid false, within: false```
2025-03-07 12:46:49 [16,684ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : decrement: 167690, decrement size: 7433845248
2025-03-07 12:46:49 [16,684ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : New limit 9574251 (slope: 447, intercept: 13179904)
2025-03-07 12:46:49 [16,685ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit buffer size 4287216384
2025-03-07 12:46:49 [16,685ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : valid true, within: true
2025-03-07 12:46:53 [20,357ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults-transient/meshlights/forceDisable'

RandomOakForest, 2025-03-12:
Thank you for posting this. [The requirements and ideal setup for Isaac Sim is listed in this doc](https://docs.isaacsim.omniverse.nvidia.com/latest/installation/requirements.html).

RandomOakForest, 2025-03-14:
Thank you for following up. You may want to try if for your specific task there are [optimizations that you can use in the Isaac Sim side](https://docs.isaacsim.omniverse.nvidia.com/latest/reference_material/sim_performance_optimization_handbook.html#isaac-sim-performance-optimization-handbook). This is actually an Isaac Sim issue and you may find other pointers in the [Isaac Sim forums](https://forums.developer.nvidia.com/c/omniverse/simulation/69).

exat500g, 2025-03-30:
我也这样，然后看见报错，软件打开的时候要从amazonaws下载素材，如果网络连接不好就会一直卡住

closed 2025-04-07 by RandomOakForest

changhoji, 2025-04-15:
I waited around 15 minutes without force-quitting, and eventually the GUI finished loading and started using the GPU properly. After that, subsequent launches were much faster (under a minute). Hope this helps others who might run into the same issue.