isaac-sim/IsaacLab #2037: [Question] Creating higher resolution rigid body meshes through the mesh converter

ethan-allan, 2025-03-07:
### Question

Hi there,

I'm currently using the mesh_converter.py script to convert STL files to USD meshes, according to this guide: https://isaac-sim.github.io/IsaacLab/v1.4.1/source/how-to/import_new_asset.html. 

The meshes I generate from here are extremely low fidelity, even when using convex decomposition. I'm wondering if there is a way to add a higher number of hulls or to generate SDF meshes using this tool.

Thanks

Mayankm96, 2025-03-07:
You can open the asset file in the UI and modify those parameters. By default, we don't support this right now.

ethan-allan, 2025-03-10:
Thanks for the reply. I'm working with the default prim structure provided by the mesh_converter.py script:

asset (defaultPrim)
----geometry
--------Looks
------------DefaultMaterial
--------mesh

The Looks and mesh prims were greyed out initially however I deselected the "make instanceable" attribute and can now edit them. I successsfully applied a SDF mesh however whenever I select "make instanceable" the mesh reverts to a convex decomposition.

Could you provide steps on editing these properties ?

Thanks

RandomOakForest, 2025-04-07:
Following up, if you still have this issue with the latest Isaac Lab release (2.0.2), please open a _Discussion_ so we can follow up. Thank you for your interest in Isaac Lab.

closed 2025-04-07 by RandomOakForest