isaac-sim/IsaacLab #2038: [Question] Control IK Franka Panda no reach the desired position

AntonioClaudiossf, 2025-03-07:
### Question

I’m trying to control the Franka Panda robot with the IK control using the run_diff_ik.py script as a base. I’m randomizing the initial values of the articulations and the desired End-Effector values. 

When I run the simulation I realize that sometimes the robot can not go to the desired position and orientation, even being within the working space of the robot.

I identified one of these cases to take the values and reproduce it. The following shows the initial values of the joints and the desired value for the End-Effector.

#### Init joints position
```
if args_cli.robot == "franka_panda":
        robot = FRANKA_PANDA_HIGH_PD_CFG.replace(prim_path="{ENV_REGEX_NS}/Robot")
        robot.init_state.joint_pos["panda_joint1"] = -0.4781
        robot.init_state.joint_pos["panda_joint2"] = -0.8056
        robot.init_state.joint_pos["panda_joint3"] = -0.3814
        robot.init_state.joint_pos["panda_joint4"] = -2.3850
        robot.init_state.joint_pos["panda_joint5"] =  0.2325
        robot.init_state.joint_pos["panda_joint6"] =  3.4186
        robot.init_state.joint_pos["panda_joint7"] =  0.5198
```
#### Desired End-Effector position and orientation
```
ee_goals = [[0.6206,  0.0244,  0.1662, 0.0000,  0.4635,  0.8860,  0.0000] ]
```
#### Initial robot position
[image]

#### Final robot position

[image]

When I start with different joint values and the same desired value for the end-effector, the robot goes to the desired position.

Could someone give me some idea of how I can solve this problem ?

Thanks ! :)

ozhanozen, 2025-03-07:
Could you monitor the joint states when this happens? Maybe they hit the joint limits.

Since the robot is kinematically redundant, I would assume the initial state affects the joint state solution corresponding to the desired end-effector pose. Some solutions could be outside the joint limits even if the end-effector is inside the workspace.

AntonioClaudiossf, 2025-03-07:
Hi @ozhanozen ! Thank you for answering!

I monitored the final values of the joints, the calculated target values using the IK controller and the joint limits, as shown below:

```
final_joint_pos  : [-0.2188,  1.4499,  1.2168, -0.5899,  1.9251,  3.7525,  2.0813]

joint_pos_des    : [ 0.2244,  1.4570,  1.2533, -0.5942,  1.8933,  4.6314,  2.0874]

joint_limits     : [[-2.8973,  2.8973],
                    [-1.7628,  1.7628],
                    [-2.8973,  2.8973],
         	    [-3.0718, -0.0698],
                    [-2.8973,  2.8973],
         	    [-0.0175,  3.7525],
         	    [-2.8973,  2.8973]]
```
We can see that for joint_6 the value of joint_pos_des "4.6314" is out of limits. :(

ozhanozen, 2025-03-07:
That could be the problem then. You can temporarily disable the joint limits to see if the robot will reach there to be sure.

As for the solution, I guess you can do different things depending on your needs and requirements:

1. Remove the joint limits and use them like that.
2. Use a robot that is not kinematically redundant. In this case, though, it is still possible that no valid solutions exist.
3. At reset, you can perform the IK operations and see if the solution found fits within the joint limits; if not, shuffle the initial joint states (or the end-effector pose) until it does.
4. Maybe try alternative methods supported for computing the inverse of the Jacobian within the `DifferentialIKController`.
5. You can also try to modify the `DifferentialIKController` to handle the nullspace as a secondary task (e.g., using the nullspace of the Jacobian).
6. Use the `OperationalSpaceController` instead; set the `nullspace_joint_pos_target` to be the center of the joint limits. This way, you are less likely to hit the limits.

I hope that this helps.

AntonioClaudiossf, 2025-03-09:
Hi @ozhanozen ! 

You were right about the limits, when I removed the robot reached the desired position (even if it is not the best behavior :( ).

[image]

Thank you for your suggestions for resolution. I will think about which one I will try to implement.

closed 2025-04-07 by RandomOakForest