isaac-sim/IsaacLab #2042: [Question] Definition of policy model using proprioceptive and visual information

YannickP99, 2025-03-08:
Hey,
I would like to implement a policy for a quadruped robot utilizing the robot's proprioceptive information as well as visual information from a camera. For this I need to define convolutional layers which process the images. The embedded image is then concatinated with the proprioceptive information of the robot.
Is it possible to implement something like this in Isaac Lab? In the examples I only found which use proprioceptive or visual information but no combination of them.
Here is a screenshot I took from a paper:
x_t: proprioceptive information
d_t: depth image

[image]

Toni-SM, 2025-03-11:
Hi @YannickP99 

You can take a look at this project: [Spaces showcase tasks for Isaac Lab
](https://github.com/Toni-SM/isaaclab-spaces-showcase)
Particularly to the [Cartpole-Camera Dict - Box configuration](https://github.com/Toni-SM/isaaclab-spaces-showcase/blob/main/source/spaces_showcase/spaces_showcase/tasks/cartpole_camera/agents/skrl_dict_box_ppo_cfg.yaml)

YannickP99, 2025-03-12:
Thank you for the help @Toni-SM, looks great

YannickP99, 2025-03-12:
I have one follow-up question. You are using the skrl model instantiator. As I understand, it does not support LSTM or GRU layers. Is it possible to define the network in another way which supports those layers?

Toni-SM, 2025-03-12:
Hi @YannickP99 

The support for LSTM or GRU layers via Isaac Lab `.yaml` config files will be added in next *skrl* minor version?

Meanwhile, you can use the [skrl's standalone Python scripts for Isaac Lab](https://skrl.readthedocs.io/en/latest/intro/examples.html#isaac-lab-environments) to use [RNN, GRU and LSTM](https://skrl.readthedocs.io/en/latest/api/models/gaussian.html#usage) with MLP and CNN layers.
You can see an example of using RNN, GRU and LSTM in the [skrl's Example section for Gymnasium](https://skrl.readthedocs.io/en/latest/intro/examples.html#gymnasium-gym-environments).

YannickP99, 2025-03-13:
Thank you for your help

closed 2025-03-13 by YannickP99