isaac-sim/IsaacLab #2043: [Question] task-space controller Jacobian matrix exception

MrVBian, 2025-03-08:
### 🔗 Linked Resources
- **Related Discussions**: [task-space controller Jacobian matrix exception #2033](https://github.com/isaac-sim/IsaacLab/discussions/2033)
- **Official Tutorial**: [Differential IK Tutorial](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/05_controllers/run_diff_ik.html)

---

### 🐛 Core Issue Description

##### Environment

   - IsaacLab: v2.0.2

[image]

In **floating-base custom robot**, the Jacobian matrix exhibits anomalies (first 4 columns are all zeros):

```log
jacobian: tensor([[
         [ 0.0000e+00,  0.0000e+00,  0.0000e+00,  0.0000e+00,  8.7464e-01, -3.1228e-02, -4.7513e-01],
         [ 0.0000e+00,  0.0000e+00,  0.0000e+00,  0.0000e+00, -2.7196e-02, 2.2162e-04, -2.5771e-02],
         [ 0.0000e+00,  0.0000e+00,  0.0000e+00,  0.0000e+00, -2.6679e-02, 2.1953e-02,  1.6447e-02],
         [ 0.0000e+00,  0.0000e+00,  0.0000e+00,  0.0000e+00, -3.0482e-02, -7.0937e-03,  3.4608e-02],
         [ 0.0000e+00,  0.0000e+00,  0.0000e+00,  0.0000e+00,  2.0839e-04, -9.9997e-01, -2.4470e-04],
         [ 0.0000e+00,  0.0000e+00,  0.0000e+00,  0.0000e+00, -9.9954e-01, 4.4119e-06,  9.9940e-01]]], device='cuda:0')
```

### 🔍 Differences (vs. Official Example)

No modifications to code logic; only the robot was replaced.

https://github.com/isaac-sim/IsaacLab/discussions/2033#discussion-8046468

**Official Tutorial**: [Differential IK Tutorial](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/05_controllers/run_diff_ik.html)

------

### ✅ Completed Verification Steps

1. **Array Index Verification**
2. **Data Checks**
   - Cartesian target reachable ✅
   - Joint control normal ✅

MrVBian, 2025-03-13:
sim_utils.ArticulationRootPropertiesCfg set is not reasonable

closed 2025-03-13 by MrVBian

RandomOakForest, 2025-03-14:
Thanks for posting this. Feel free to reopen the issue may you need further help.