isaac-sim/IsaacLab #2050: [Question] List of available assets

sdfzz, 2025-03-09:
Hi,

Is there a way to obtain the list of available assets in IsaacLab?

My problem is that I want to use 23 DOF Unitree G1 for my project, but G1 in "Isaac-Velocity-Rough-G1-v0" has 43 DOF.

I was checking "unitree.py" in lab_assets, there is no list of available robots.

Any help would be greatly appreciated

Regards,

Steve

celestialdr4g0n, 2025-03-10:
Hi, 
You can try to download the asset using the link given in eny.yaml config file in /logs, for example logs/skrl/shadow_hand_over/.../params/env.yaml. Then search for ".usd", the link should be there.
Another way is to open Isaac Sim and download the robots that you want.

sdfzz, 2025-03-10:
Hi,
Thank you for the reply. You are right, I guess the best option is to download robots from Isaac Sim.
Regards,
Steve

closed 2025-03-10 by sdfzz