isaac-sim/IsaacLab #2053: [Bug Report] Receive warnings when visualization contact states in Isaac Lab

xinyu-66, 2025-03-10:
### Describe the bug

When I running the training with rl-games package including the visulization of contact, there are two warnings reporting over and over again when contact happens.

### Steps to reproduce

I also observe it when running the contact sensor tutorial:
https://isaac-sim.github.io/IsaacLab/main/source/overview/core-concepts/sensors/contact_sensor.html

warning img:
file:///home/xinl/Downloads/Screenshot%20from%202025-03-10%2012-59-05.png

warning text:
2025-03-10 11:57:01 [190,863ms] [Warning] [omni.usd] Warning (secondary thread): in GetInstanceIndices at line 2190 of /builds/omniverse/usd-ci/USD/pxr/usdImaging/usdImaging/pointInstancerAdapter.cpp -- ProtoIndex 1 out of bounds (prototypes size = 1) for (/Visuals/ContactSensor, /Visuals/ContactSensor.proto1_mesh_id0) 

2025-03-10 11:57:01 [190,863ms] [Warning] [omni.usd] Warning (secondary thread): in GetInstanceIndices at line 3256 of /builds/omniverse/usd-ci/USD/pxr/usdImaging/usdImaging/delegate.cpp -- Empty InstanceIndices (/Visuals/ContactSensor, /Visuals/ContactSensor.proto1_mesh_id0) 

### System Info

Describe the characteristic of your environment:

- Commit: []
- Isaac Sim Version: [4.5.0]
- OS: [Ubuntu 22.04]
- GPU: [RTX 4090]
- CUDA: [e.g. 12.1]
- GPU Driver: [e.g. 573.82.01, this can be seen by using `nvidia-smi` command.]

### Additional context

Add any other context about the problem here:

It seems has no influence on training but just in case something wrong.

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [X] if this warning needs to be solved or just let it go?

RandomOakForest, 2025-03-13:
Thank you for catching this. The team will review.

xinyu-66, 2025-03-14:
> Thank you for catching this. The team will review.

Thanks! Just

xinyu-66, 2025-03-14:
sorry, I just want to say thanks but misclick close issue

closed 2025-03-14 by xinyu-66

hojae-io, 2025-04-15:
I also have the same issue, mine with `IsaacLab/scripts/demos/sensors/contact_sensor.py`

```
2025-04-15 04:42:19 [18,905ms] [Warning] [omni.usd] Warning (secondary thread): in GetInstanceIndices at line 2190 of /builds/omniverse/usd-ci/USD/pxr/usdImaging/usdImaging/pointInstancerAdapter.cpp -- ProtoIndex 1 out of bounds (prototypes size = 1) for (/Visuals/ContactSensor, /Visuals/ContactSensor.proto1_mesh_id0)

2025-04-15 04:42:19 [18,905ms] [Warning] [omni.usd] Warning (secondary thread): in GetInstanceIndices at line 2190 of /builds/omniverse/usd-ci/USD/pxr/usdImaging/usdImaging/pointInstancerAdapter.cpp -- ProtoIndex 1 out of bounds (prototypes size = 1) for (/Visuals/ContactSensor_01, /Visuals/ContactSensor_01.proto1_mesh_id0)

2025-04-15 04:42:19 [18,905ms] [Warning] [omni.usd] Warning (secondary thread): in GetInstanceIndices at line 3256 of /builds/omniverse/usd-ci/USD/pxr/usdImaging/usdImaging/delegate.cpp -- Empty InstanceIndices (/Visuals/ContactSensor, /Visuals/ContactSensor.proto1_mesh_id0)

2025-04-15 04:42:19 [18,905ms] [Warning] [omni.usd] Warning (secondary thread): in GetInstanceIndices at line 3256 of /builds/omniverse/usd-ci/USD/pxr/usdImaging/usdImaging/delegate.cpp -- Empty InstanceIndices (/Visuals/ContactSensor_01, /Visuals/ContactSensor_01.proto1_mesh_id0)
```

Has this been resolved?

kellyguo11, 2025-07-05:
the warning is safe to ignore, we are looking into the cause for it but it seems to point to USD for visualization, so it should not impact simulation

kellyguo11, 2025-07-05:
I believe these may already be fixed in Isaac Sim 5.0. If you'd like to try the OSS isaac sim branch (https://github.com/isaac-sim/IsaacSim) with the `feature/isaacsim_5_0` branch of lab, the warnings may already be gone.