isaac-sim/IsaacLab #2054: [Bug Report] robot.data.joint_effort_limits is not correctly set for implicit actuators

Hellod035, 2025-03-10:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug
The effort limit in usd is correct
[image]
but robot.data.joint_effort_limits is 1e9. I'm using ImplicitActuator.

### Steps to reproduce

### System Info

Describe the characteristic of your environment:

- Commit: [Latest]
- Isaac Sim Version: [4.5.]
- OS: [Ubuntu 22.04]
- GPU: [4090D]
- CUDA: [12.4]
- GPU Driver: [550.144.03]

### Additional context

Add any other context about the problem here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-03-13:
Thank you for posting this. There may be an initial state before simulation begins. When are you printing the value of `robot.data.joint_effort_limits`? Could you share code to reproduce this? Thanks.

Hellod035, 2025-03-13:
This happens for any asset as long as `effort_limit_sim` or `effort_limit` is not set. In IsaacLab,  I commented out `effort_limit` in unitree g1 assets and test
```
 {
            "name": "Python: Train Environment",
            "type": "python",
            "request": "launch",
            "args" : ["--task", "Isaac-Velocity-Flat-G1-v0", "--headless", "--num_envs=64"],
            "program": "${workspaceFolder}/scripts/reinforcement_learning/rsl_rl/train.py",
            "console": "integratedTerminal"
 }
```

The same thing happened in [legged lab](https://github.com/Hellod035/Legged_Lab). I commented out `effort_limit_sim` in unitree g1 assets and test
```
        {
            "name": "Train h1_flat",
            "type": "debugpy",
            "request": "launch",
            "program": "${workspaceFolder}/legged_lab/scripts/train.py",
            "args": [
                "--task=g1_flat",
                "--headless",
                "--logger=tensorboard",
                "--num_envs=64"
            ],
        }
```
in  [legged lab](https://github.com/Hellod035/Legged_Lab).

`asset.data.joint_effort_limits` is set to1e9 even though it seems that the USD assets visualized in isaacsim have a max force cap.

WangYCheng23, 2025-03-14:
Similarly for me, it doesn't read the effort limit from IsaacSim, but it does read the velocity limit. So, you'll need to write that in <your robot>.py file. :)

Hellod035, 2025-03-17:
> Thank you for posting this. There may be an initial state before simulation begins. When are you printing the value of `robot.data.joint_effort_limits`? Could you share code to reproduce this? Thanks.

This is actually a serious bug that needs to be fixed immediately!
The joint torque limits specified in the asset are not inherited in the ImplicitActuator!

PR #2098 (2025-03-19): Fixes default effort limit behavior for implicit actuators
# Description

Fix default behavior of implicit actuators if no effort limit is set.

Fixes #2054 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Screenshots

Please attach before and after screenshots of the change if applicable.

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with `./isaaclab.sh --format`
- [x] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
- [ ] I have added tests that prove my fix is effective or that  [... cut]
commit d7da02da (2025-03-19): Fixes default effort limit behavior for implicit actuators (#2098)

# Description

This MR fixes the default behavior of implicit actuators if no effort limit is set. Previously, the check
was using the class variable value instead of self which led to the wrong values getting propogated.

Fixes #2054 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.sh --format`
- [x] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
- [ ] I [... cut]
closed 2025-03-19 by Mayankm96